analogdevicesinc/libm2k #1: OS X Build Problem with Exceptions

jnlaneman, 2019-06-27:
I am trying to build libm2k from the master repo on Mac OS X 10.14.5, and there appears to be a problem with exception handling in the include files. Recurring error of "use of undeclared identifier '__try'"

For context, I have installed libiio via MacPorts (https://github.com/macports/macports-ports/blob/master/science/libiio/Portfile).

My command log and output are below.

[JNL-ND-iMac:~] jnl% uname -a
Darwin JNL-ND-iMac.local 18.6.0 Darwin Kernel Version 18.6.0: Thu Apr 25 23:16:27 PDT 2019; root:xnu-4903.261.4~2/RELEASE_X86_64 x86_64
[JNL-ND-iMac:~] jnl% git clone https://github.com/analogdevicesinc/libm2k.git
Cloning into 'libm2k'...
remote: Enumerating objects: 43, done.
remote: Counting objects: 100% (43/43), done.
remote: Compressing objects: 100% (36/36), done.
remote: Total 2225 (delta 14), reused 20 (delta 6), pack-reused 2182
Receiving objects: 100% (2225/2225), 699.90 KiB | 3.17 MiB/s, done.
Resolving deltas: 100% (1498/1498), done.
[JNL-ND-iMac:~] jnl% cd libm2k
[JNL-ND-iMac:~/libm2k] jnl% cmake ./
-- The CXX compiler identification is AppleClang 10.0.1.10010046
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.20.1 (Apple Git-117)") 
-- Found SWIG: /opt/local/bin/swig (found version "3.0.12") 
---- Building Python bindings
-- Found PythonLibs: /opt/local/lib/libpython2.7.dylib (found version "2.7.10") 
-- Found PythonInterp: /opt/local/bin/python2.7 (found version "2.7.16") 
CMake Warning (dev) at /opt/local/share/cmake-3.14/Modules/UseSWIG.cmake:600 (message):
  Policy CMP0078 is not set: UseSWIG generates standard target names.  Run
  "cmake --help-policy CMP0078" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

Call Stack (most recent call first):
  bindings/python/CMakeLists.txt:23 (SWIG_ADD_LIBRARY)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /opt/local/share/cmake-3.14/Modules/UseSWIG.cmake:460 (message):
  Policy CMP0086 is not set: UseSWIG honors SWIG_MODULE_NAME via -module
  flag.  Run "cmake --help-policy CMP0086" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /opt/local/share/cmake-3.14/Modules/UseSWIG.cmake:695 (SWIG_ADD_SOURCE_TO_MODULE)
  bindings/python/CMakeLists.txt:23 (SWIG_ADD_LIBRARY)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /Users/jnl/libm2k
[JNL-ND-iMac:~/libm2k] jnl% make all
Scanning dependencies of target libm2k
[  1%] Building CXX object src/CMakeFiles/libm2k.dir/analog/dmm.cpp.o
[  3%] Building CXX object src/CMakeFiles/libm2k.dir/analog/genericanalogin.cpp.o
In file included from /Users/jnl/libm2k/src/analog/genericanalogin.cpp:20:
/Users/jnl/libm2k/src/analog/private/genericanalogin_impl.cpp:51:35: warning: 
      'libm2k::analog::GenericAnalogIn::GenericAnalogInImpl::getSamples' hides
      overloaded virtual functions [-Woverloaded-virtual]
        std::vector<std::vector<double>> getSamples(unsigned int nb_samples,
                                         ^
/Users/jnl/libm2k/include/libm2k/utils/devicein.hpp:44:38: note: hidden
      overloaded virtual function 'libm2k::utils::DeviceIn::getSamples' declared
      here: different number of parameters (1 vs 2)
        virtual std::vector<unsigned short> getSamples(unsigned int nb_samples);
                                            ^
/Users/jnl/libm2k/include/libm2k/utils/devicein.hpp:45:44: note: hidden
      overloaded virtual function 'libm2k::utils::DeviceIn::getSamples' declared
      here: type mismatch at 2nd parameter ('std::function<double
[... 71 lines cut ...]
                ^
/Users/jnl/libm2k/src/analog/private/m2kanalogin_impl.cpp:372:3: error: use of
      undeclared identifier '__try'
                __try {
                ^
/Users/jnl/libm2k/src/analog/private/m2kanalogin_impl.cpp:389:3: error: use of
      undeclared identifier '__try'
                __try {
                ^
/Users/jnl/libm2k/src/analog/private/m2kanalogin_impl.cpp:405:3: error: use of
      undeclared identifier '__try'
                __try {
                ^
/Users/jnl/libm2k/src/analog/private/m2kanalogin_impl.cpp:426:3: error: use of
      undeclared identifier '__try'
                __try {
                ^
/Users/jnl/libm2k/src/analog/private/m2kanalogin_impl.cpp:437:3: error: use of
      undeclared identifier '__try'
                __try {
                ^
1 warning and 6 errors generated.
make[2]: *** [src/CMakeFiles/libm2k.dir/analog/m2kanalogin.cpp.o] Error 1
make[1]: *** [src/CMakeFiles/libm2k.dir/all] Error 2
make: *** [all] Error 2

AlexandraTrifan, 2019-06-28:
Hi,

Unfortunately, we didn't test the build on Mac OS X yet. We'll look into it, make the necessary changes and get back to you with some info as soon as possible.

Thank you.

-Alexandra

jnlaneman, 2019-06-28:
Thanks, Alexandra.

We are actively using the m2k to develop a course for Spring 2020. Two graduate students, an undergraduate student, and myself. And we all have Macs in addition to several Linux PCs.

If we can be of some assistance, we are happy to do so. I’ve also bugged my former PhD student, because he apparently did the MacPort for LibIIO.

AlexandraTrifan, 2019-07-15:
Hi, 

The problem should be fixed after this commit: 7a498ade29dd7b106872372e2cdf5d2929d58fb5 .
Could you try it again? You should clean the build folder first

Thank you.

-Alexandra

AlexandraTrifan, 2019-10-15:
@jnlaneman Any updates or issues?

AlexandraTrifan, 2019-10-28:
Closing. Can be reopened if anything new appears.

closed 2019-10-28 by AlexandraTrifan