analogdevicesinc/libm2k #3: Add command-line tools

AlexandraTrifan, 2019-07-01:
Add M2K examples that will be deployed with the library and can be used as command-line tools such as: 

- Setup the ADC and capture samples: Example: ./m2k_capture <samplerate> <trigger y/n>  
- Setup the DAC and push samples: Example: ./m2k_generate < samples_file.csv
- Similar for the digital part.
- Calibrate ADC/DAC
- Get voltage 

Each tool should be designed with the appropriate parameters. 
This list should be expanded.

rgetz, 2019-07-01:
Are these examples going to be python, C or C++?

There are more people asking for python examples than anything... (from what I see).

AlexandraTrifan, 2019-07-02:
These CLI tools should be some executables, we will create them as C++ examples. But we will also add some more Python examples.

teoperisanu, 2019-10-15:
PR merged #38

closed 2019-10-15 by teoperisanu