analogdevicesinc/libm2k #11: What's going wrong here?

damercer, 2019-08-01:
Hi:
As you will see in error captured from the Python shell, I'm running 3.7.3 on a 64 bit Windows 7 computer. I installed libm2k for windows using libm2k-system-setup.exe from this Appveyor build:

https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/26172407/artifacts

I downloaded the simplest example powersupply.py and tried to run it but it failed???

Python 3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 22:22:05) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license()" for more information.
>>> 
==================== RESTART: C:\ALICEm2k\powersupply.py ====================
Traceback (most recent call last):
  File "C:\ALICEm2k\powersupply.py", line 1, in <module>
    import libm2k
  File "C:\Python37\lib\site-packages\libm2k.py", line 15, in <module>
    import _libm2k
ImportError: DLL load failed: %1 is not a valid Win32 application.
>>> 

So my question. What's going wrong here? What's missing in this installation?

Thanks

Doug

AlexandraTrifan, 2019-08-05:
Hi, 

Can't reproduce it on Windows 10 with the same Python version. 
I will look for a Windows 7 machine to test and get back with a fix.

-Alexandra

damercer, 2019-08-05:
Hi:
Thanks for checking into this. I think there might be more to this problem in that I can't get libiio or libsmu to install and work with this new Python 3.7 either. All three return more or less the same error message when trying to import the bindings.

Doug

AlexandraTrifan, 2019-08-05:
Hm, that's weird. Let me check and I'll get back to you with some answers.

-Alexandra

rgetz, 2019-08-05:
@damercer :

First thing to check: is libiio installed and working. From a DOS/Command prompt, check out:
```
iio_info -s
iio_attr -a -C
```

the first one should give you a scan:
```
C:\Users\rgetz>iio_info -s
Library version: 0.17 (git tag: 5bdc242)
Compiled with backends: xml ip usb serial
Available contexts:
        0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=10440004278a0007f5ff1e00ed95edf378 [usb:1.6.5]
```

the second should tell you what device you are connected to:
```
C:\Users\rgetz>iio_attr -a -C
Using auto-detected IIO context at URI "usb:1.6.5"
WARNING: Skipping broken USB device. Please upgrade libusb.
IIO context with 13 attributes:
hw_model: Analog Devices PlutoSDR Rev.B (Z7010-AD9363)
hw_model_variant: 1
hw_serial: 10440004278a0007f5ff1e00ed95edf378
fw_version: v0.26
ad9361-phy,xo_correction: 40000000
ad9361-phy,model: ad9363a
local,kernel: 4.9.0-10194-g40c2158
usb,idVendor: 0456
usb,idProduct: b673
usb,release: 2.0
usb,vendor: Analog Devices Inc.
usb,product: PlutoSDR (ADALM-PLUTO)
usb,serial: 10440004278a0007f5ff1e00ed95edf378
```

If those work - I (or Travis) can give you some Python to do the same.

tfcollins, 2019-08-05:
Python equivalent (ish) for libiio assuming you have the bindings:
 ```
C:\Users\tcollins>python -c "import iio; ctxs = iio.scan_contexts(); print(ctxs)"
{'usb:1.5.5': '0456:b672 (Analog Devices Inc. M2k (ADALM-2000)), serial=104473ce699100011100360064018a880f'}
```
Python equivalent for libm2k
```
C:\Users\tcollins>python -c "import libm2k; ctx = libm2k.m2kOpen(); print(ctx.getUri())"
M2K
usb:1.5.5
```

damercer, 2019-08-05:
As far as libiio goes I think it is installed because it is still ( and always had ) working using Python 2.7 and the binding for 2.7. But now for 3.7 not so much. I then re-ran the installer and there were only check boxes for older version of Python 3. Then today I tried the installer again and there was a check box for Python 3.7. I checked that and it then confirmed the Python37 directory and finished. I opened a Python 3.7 shell and typed import iio and it came back with:

Python 3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 22:22:05) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license()" for more information.
>>> import iio
Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    import iio
  File "C:\Python37\lib\site-packages\iio.py", line 62, in <module>
    use_errno = True, use_last_error = True)
  File "C:\Python37\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application
>>> 
So there is an iio.py in site packages so the installer did something?

Much like the libm2k error message.

tfcollins, 2019-08-05:
I would first remove all versions of libiio, libm2k, and their bindings using Add/Remove programs. Download the latest libm2k installer and try to import the module. Running the same command above for libm2k.

damercer, 2019-08-05:
Nope that didn't help. Removed everything (except Python 3.7) and first re-ran libm2k installer. Installer had check boxes for libm2k and libiio. Get same error in shell for import libm2k.
>>> import libm2k
Traceback (most recent call last):
  File "<pyshell#1>", line 1, in <module>
    import libm2k
  File "C:\Python37\lib\site-packages\libm2k.py", line 15, in <module>
    import _libm2k
ImportError: DLL load failed: %1 is not a valid Win32 application.
>>> 
 import iio returned no iio.py found. 
Re-ran libiio installer. import iio in shell returns 

>>> import iio
Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    import iio
  File "C:\Python37\lib\site-packages\iio.py", line 62, in <module>
    use_errno = True, use_last_error = True)
  File "C:\Python37\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application

damercer, 2019-08-05:
iio_info.exe and iio_attr.exe both in Windows/System32 exist and are dated 7/26/2019.
With an m2k plugged in both return nothing when run from a mingw32 shell. When run from an Windows command prompt an error window pops up.

[image]

But once again I'm puzzled as to why iio still works in my Python 2.7 (ALICE for m2k) program???

tfcollins, 2019-08-05:
I'm guessing the compilers we are using now might not support Windows 7 anymore. I don't have a Windows 7 machine but maybe Cluj has one

damercer, 2019-08-06:
Might it have something to do with 64 bit vs 32 bit versions? The reason I ask is that my working Python 2.7 is 32 bit version (for various historical reasons) and this new not working Python 3.7 is  64 bit. Can both 32 bit and 64 bit versions of libiio exist on the same computer?

Doug

damercer, 2019-08-10:
I have some further details on attempting to make progress on this issue:
I removed the (AMD)64 bit version of Python 3.7 and used the default Windows installer for Python 3.7.4 which is 32 bit. I then used pip to install numpy and that worked fine. I installed libiio and my program that uses libiio started working now.

I then used the same libm2k installer as above. Now when I run the installer I un-check the box for installing libiio (already there) and with just the install for Python 3.7 checked I get this error message:

[image]

It does not let me manually enter the directory for where Python is installed.

Seems strange because Python 3.7 is listed in the Start>Programs list and I have the option to edit .py files with the Python IDLE for 3.7? I had all the boxes checked when installing Python.

Any additional thoughts on fixing this issue would be great.

Thanks

Doug

AlexandraTrifan, 2019-08-12:
Hi, 

I tried to install Python using the default Windows installer for 3.7.4 - the 32 bit one. When I get to the point where I can select the Python version (for both the libiio and libm2k installer) , it does not detect the Python3.7 version (because that one is a 32-bit version, while the installer searches for the 64-bit version) . 

After that, I downloaded the Python 3.7.4 - Windows x86-64 executable installer ( the link here: https://www.python.org/downloads/release/python-374/ ) . I checked all the boxes when installing, added Python to the Path. Then I installed libiio (using the nightly build installer) and tested the Python bindings, they worked fine. I did the same for the libm2k installer, I got to the Python bindings installer and now 3.7 was listed there, I chose it, it pointed to the right location and installed properly. 

I think the general libm2k installer checks if the system is 32-bit or 64-bit and selects the library and the corresponding Python bindings. I this case, the 64-bit Python libm2k executable was running, but it could not find a 64-bit Python version.

I will work on creating and distributing .whl files for libm2k Python bindings, this way you should be able to install on both Python 64-bit and 32-bit using pip. Until those are done, there are a few things you could try: 

1. Install Python 3.7.4 using the installer from the link I provided above. That should get a Python 64-bit (probably located in C:\Program Files\Python37 ). The libm2k installer detects that version and installs the bindings properly. 

2. Copy the bindings manually in you Python 3.7.4 32-bit version. 
If you want to use that version, you need install libm2k without Python bindings box checked, and copy 2 files for the libm2k bindings.
In the nightly build of libm2k, there are 2 zip files: https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/26530546/artifacts . Get the 32-bit zip file, unzip it. From that archive copy **_libm2k.pyd** and **libm2k.py** in the **C:\Program Files(x86)\Python37-32\Lib\site-packages** (I assumed that's the location of your Python 32-bit. ) 

Please let me know if any of these works for you.

-Alexandra

damercer, 2019-08-12:
Hi:
Copied those two files from the archive zip file in to site-packages. The powersupply.py example program now works. I might give the 64 bit version of Python another try when I have more time. I used this installer, python-3.7.3-amd64.exe, in the earlier attempt vs the one you suggested, Not sure that would have made a difference or not.

So for now I think I can proceed while you get the rest of the installer process sorted out for 32 bit systems.

Thanks

Doug

AlexandraTrifan, 2019-08-13:
I will add some fixes for the installer as soon a possible. 
Please let us know if there are any more issues.

-Alexandra

AlexandraTrifan, 2019-12-20:
Hi,

In the artifacts there now are 2 .whl files. If you have a 64-bit machine, the installer will always try to install Python bindings for your machine. But if you do not select that option, you can download .whl files from the latest artifacts from Appveyor. Those can be installed with pip like `pip install libm2k-0.1.0-py3-none-win32.whl` and will install the correct bindings for Python 32-bit.

-Alexandra

adisuciu, 2020-07-08:
I'm closing this for inactivity, reopen if the issue is still present.
-Adrian

closed 2020-07-08 by adisuciu