analogdevicesinc/libm2k #12: How do I set or change the scope input hardware offset?

damercer, 2019-08-13:
Hi:

I'm going through the various things I used in libiio to control the m2k and finding the corresponding function in libm2k. I'm finding a lot of them but have not found how to set or control the analog input channel hardware DC offset, used to change the vertical position. This was done in libiio by changing the value sent to the ad5625 quad DAC as follows:

ad5625 = ctx.find_device("ad5625")
Scope1Offset = ad5625.find_channel("voltage2", True)
Scope2Offset = ad5625.find_channel("voltage3", True)
Scope1Offset.attrs["raw"].value = '2009'
Scope2Offset.attrs["raw"].value = '2028'

Changing the DC offset or position is a very important function which allows for example, in the high gain range setting the nominal -2.5 V to 2.5 V span to be shifted to 0 to +5 V.

So the question is what function in the API is used to do this?

Thanks

Doug

AlexandraTrifan, 2019-08-14:
Hi, 

This was added later on. Right now, it's on this branch: https://github.com/analogdevicesinc/libm2k/tree/fixes . I will check the additions once again and soon merge it into master. 

Thank you,

-Alexandra

AlexandraTrifan, 2019-08-14:
Fixes were merged to master. The Analog In class exposes a setter and getter for vertical offset.

-Alexandra

damercer, 2019-08-14:
Great, thanks. Where is this (will be) documented? Has it made it into: https://analogdevicesinc.github.io/libm2k/modules.html ?
Also I've been looking around here under libm2k for the python source code libm2k.py or whatever and only see the examples under the Python Bindings?

Doug

AlexandraTrifan, 2019-08-19:
Hi, 

The documentation was not updated yet. We will do that as soon as possible.

The libm2k.py file can't be found in the source code. The Python code is automatically generated on every build, using SWIG,  from this file: https://github.com/analogdevicesinc/libm2k/blob/master/bindings/libm2k.i .

-Alexandra

AlexandraTrifan, 2019-10-15:
I assume this is fixed, I am closing it. Can be reopened if any issue appears.

closed 2019-10-15 by AlexandraTrifan