analogdevicesinc/libm2k #13: Extracting values from SwigPyObject?

damercer, 2019-08-13:
Hi:
Many of the libm2k functions return a SwigPyObject like
ain.getRangeLimits(0) returns:
<Swig Object of type 'std::pair< double,double > *' at 0x06A0E8A8>
ain.getAvailableRanges() returns
<Swig Object of type 'std::vector< libm2k::analog::M2K_RANGE,std::allocator< libm2k::analog::M2K_RANGE > > *' at 0x065DF0C8>

How do I convert or extract these Swig objects to normal python variables?

Thanks

Doug

AlexandraTrifan, 2019-08-14:
Hi, 

I added a fix for this in the latest master. It was not correctly exposed to the Python bindings.

Thanks 

-Alexandra

AlexandraTrifan, 2019-10-15:
I assume this is fixed, I am closing it. Can be reopened if any issue appears.

closed 2019-10-15 by AlexandraTrifan

codenio, 2021-07-26:
Hi, getContextsInfo returns Swig Object, unable to extract data from it. does it need correction in python binding..?

```python
> ibm2k.getContextsInfo()
<Swig Object of type 'std::vector< libm2k::CONTEXT_INFO *,std::allocator< libm2k::CONTEXT_INFO * > > *' at 0x7f90e010a480>
```

AlexandraTrifan, 2021-07-26:
Hi,

Indeed, this requires a small correction in the python bindings definition. 
I added the fix and it can be tracked here: https://github.com/analogdevicesinc/libm2k/pull/258 .
Will run a few more tests and merge it into master.

Thank you!
-Alexandra

codenio, 2021-07-27:
added the changes as per #258, build it from the source. now binding for libm2k::CONTEXT_INFO needs correction..?

```
>>> import libm2k
>>> libm2k.getContextsInfo()
<libm2k.VectorCtxInfo; proxy of <Swig Object of type 'std::vector< libm2k::CONTEXT_INFO *,std::allocator< libm2k::CONTEXT_INFO * > > *' at 0x7fe23ce6d660> >
>>> for i in libm2k.getContextsInfo():
...     print(i)
... 
<Swig Object of type 'unknown' at 0x7fe23ce6d960>
``` 

Thoughts and Comments..?

AlexandraTrifan, 2021-07-27:
Hi,

It seems that iterating through a vector filled with structs does not work fine in SWIG generated bindings.
You can still access the object in the following way:

```
import libm2k
lst = libm2k.getContextsInfo()
for i in range(len(lst)):
    print(lst[i].manufacturer)
```

I will add a new commit which will remove this limitation and will allow you to access the objects using an iterator:
```
for obj in libm2k.getContextsInfo()
       ...
```

However, printing the object itself will not work. You can access separately any of the object attributes that are available ( https://analogdevicesinc.github.io/libm2k/structlibm2k_1_1_c_o_n_t_e_x_t___i_n_f_o.html ) 

-Alexandra

codenio, 2021-07-27:
Thanks a lot, It really helps. :+1: