analogdevicesinc/libm2k #14: Question anout analog input buffer

damercer, 2019-08-15:
Hi:

In my testing so far it seems that the analog input buffer still contains some old data after I change the sample rate and maybe when I change the number of samples, SHOWsamples, in a call like this:
ain.getSamples(SHOWsamples)

Is there a way to flush out old samples after a change other than doing one or more extra ain.getSamples as dummy reads?

Thanks

Doug

AlexandraTrifan, 2019-08-29:
Hi, 

There are a couple of new commits on master now. One of them added a function which allows to change the number of kernel buffers. Could you try to run: setKernelBuffersCount(1) and tell us whether this looks the same? 

Thank you, 

-Alexandra

damercer, 2019-08-29:
Which version from AppVeyor should I download and test? The latest ones all seem to be failing before they complete?

Let me know and I can test.

Doug

damercer, 2019-08-30:
Hi:
In removed my existing libm2k with the Control Panel and downloaded this new version from AppVeyor master.118. I installed this and as before I also manually copied these two files from the new libm2k-win32.zip archive. _libm2k.pyd and libm2k.py

Now I get this error that I've not seen before:

Traceback (most recent call last):
  File "C:\ALICEm2k\alice-desktop-2.1.pyw", line 13, in <module>
    import urllib.request, urllib.error, urllib.parse
ImportError: No module named request
>>> 

Doug

damercer, 2019-08-30:
Sorry, disregard that last comment, I accidentally ran the program with Python 2.7.
But, When I do run with Python 3.7 I get this error:

Traceback (most recent call last):
  File "C:\ALICEm2k\alice-desktop-2.1.pyw", line 17525, in <module>
    ConnectDevice() # find attached M2K
  File "C:\ALICEm2k\alice-desktop-2.1.pyw", line 8187, in ConnectDevice
    trig.setSource(0) # Channel 0 as source
AttributeError: 'M2kHardwareTrigger' object has no attribute 'setSource'

So it seems something has changed. What is the new way to set the trigger source?

Thanks

Doug

AlexandraTrifan, 2019-08-30:
Hi, 

Sorry for the inconvenience. 
M2kAnalog and M2kDigital no longer contain any API functions that modify the trigger settings. All the trigger handling is done inside the M2kHardwareTrigger object. 

In order to make a difference between the specific methods, their names changed. For example: setSource -> setAnalogSource . The available methods are here: https://github.com/analogdevicesinc/libm2k/blob/master/include/libm2k/analog/m2khardwaretrigger.hpp .

We are in process of updating and merging the documentation. So we will also add the M2kHardwareTrigger class. 

Thank you.

-Alexandra

damercer, 2019-08-31:
Hi:
I added a new line with setKernelBuffersCount(1) like this where I set up everything to talk to m2k:

# definer analog inputs and outputs
    ain=ctx.getAnalogIn()
    aout=ctx.getAnalogOut()
    trig=ain.getTrigger()
    ain.setKernelBuffersCount(1) # added line here

I assume that is the correct usage.

I took out the extra dummy getSamples and the problem with old data in the buffer is still there. With the extra dummy getSamples all is good.

So that was not the the problem (solution).

Thanks

Doug

teoperisanu, 2019-09-02:
Hello Doug,

Can you share your full code ?  We ran some test code with the latest library (https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/27078023/artifacts) and we couldn't find the issue ..

The code and input file that we used can be found here - https://gist.github.com/teoperisanu/4f441ecdb3a1568f9bc4f1bee59ef432

In this example, we do the following:
- set kernel buffers to 1
- start the analog out with a sine wave 
- read back the data on the analog input and plot it.
- stop the waveform
- read analog input and plot it 
- a clear difference can be seen between the plots

If we change the number of kernel buffers  to 10 (for example), even after stopping the waveform, the acquisition is still there - and it's contigous.

-Teo

damercer, 2019-09-02:
I will attach my code but ..... My code is a rather large (17500 lines Full implementation of the ALICE desktop for m2k). I'm finding the issue in two use cases. One is when using an analog input mux (to provide an alternate sweep mode to increase the number of scope channels). In this case as the input mux address bits are changed (between "sweeps" or getSamples calls ) data from the previous mux
 input is still in the buffer and thus now appears in the wrong (next) trace. If I do a dummy read like this before the read I actually use it works

From Line 2898 in def Analog_Time_In():

if MuxScreenStatus.get() > 0: # if external mux mode dummy read to empty buffer
        ADsignal1 = ain.getSamples(SHOWsamples*2) # read the buffer
ADsignal1 = ain.getSamples(SHOWsamples) # read the buffer

Note the SHOWsamples*2 in the dummy read. If just read SHOWsamples it does not fix the problem.

To reproduce this problem requires attaching an external analog Mux IC to the channel 2 scope input and the DIO pins for the mux address. 

The second use case is in the Bode Plotter tool. At selected points in the frequency sweep as the test frequency changes, I change the FFT size (number of samples). The plotted amplitude and phase do not change as they should at these points in the sweep. If I add the dummy read as above it fixes the issue:

From line 10849 in def Analog_Freq_In():

if SHOWsamples != SMPfft:
        # print('changing number of samples')
    SHOWsamples = SMPfft
ADsignal1 = ain.getSamples(SHOWsamples*2) # do a dummy read to clear old data from buffer??
ADsignal1 = ain.getSamples(SHOWsamples)

You might be able to try to reproduce this case by trying to do a Bode plot of simple RC low pass filter. The zero stuffing control option has to be manually set to 3 in this version of the code.

This screen shot is for the case where the dummy read is commented out: Note the flat places in the orange trace.

[image]

This screen shot is with the dummy read. Note the now proper smooth orange trace.

[image]
Here is a zip of the Python code and the init and config files that should set you up for a RC LP filter bode plot (R = 1K C= 0.22u).

[alice-desktop-2.1.zip](https://github.com/analogdevicesinc/libm2k/files/3566277/alice-desktop-2.1.zip)
[ALICEm2k.zip](https://github.com/analogdevicesinc/libm2k/files/3566320/ALICEm2k.zip)

teoperisanu, 2019-09-04:
Hello Doug,

We have run some tests and it seems that the calibration process was creating a buffer without destroying it. Because of this, the number of kernel buffers couldn't be set properly. We added some fixes that should resolve this issue. Can you get the libm2k version from the 'Kernel buffer fix' branch (https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/27176993/artifacts) and set the number of kernel buffers to 1 (ain.setKernelBuffersCount(1)). Please tell us if the result is the same.

-Teo

damercer, 2019-09-04:
I installed the master.129 version since it seemed to contain the same changes as  'Kernel buffer fix' branch. With the ain.setKernelBuffersCount(1) included the results of the Bode Plotter is now correct without a dummy read. I have not tested the analog mux mode but I'm fairly sure that will work now as well. Is there any benefit (like update speed) with more than one Kernel Buffer?

Thanks

Doug

adisuciu, 2019-09-04:
If you want want your data to be continuous across multiple buffers, you need kernel buffers.

The problem is that it depends whether you are doing calibration or no. The calibration procedure can leave the board in a weird state that doesn't allow the update of kernel buffers. Things should be fixed with Teo's version.

-Adrian

damercer, 2019-09-04:
With the fix if I set ain.setKernelBuffersCount(2) rather than 1 I start getting the data problem. So I'm not following exactly what you mean but it does not really matter right now since things work when set to 1.

Doug

adisuciu, 2019-09-05:
After applying the fix, try restarting the board, as you probably calibrated(at least once) without the fix that messed things up again. If you're still having trouble we can look into it some more.

-Adrian

closed 2019-10-01 by adisuciu