analogdevicesinc/libm2k #22: Unable to build the latest version of libm2k on linux

xwkang2019, 2019-09-25:
Hey Guys,
I have been using ADALM2000 with libm2k on Windows System and the fix last month on pushing non-cyclic buffers has helped me a lot on a project I am working on. Now we need to run our programs on a Linux system, and therefore I would have to install the latest version of libm2k. However, after following the commands provided on "https://github.com/analogdevicesinc/libm2k/wiki/Building-libm2k-on-Linux" I was getting errors in building libm2k. One of the errors says "class M2K Digital has no member named 'getTriggerMode', did you mean 'getTrigger'?" I know in our latest commits some functions haven been renamed and I noticed a file named "LinuxPackaging.cmake" was 4 month old. Is that the reason that we are unable to build the latest libm2k on linux? (I am new to Linux and CMake...)

Thanks in advance for your help!
Steven

AlexandraTrifan, 2019-09-25:
Hello, 

I just tested on a clean Ubuntu 18.04 using this build guide: " https://github.com/analogdevicesinc/libm2k/wiki/Building-libm2k-on-Linux " and I could not reproduce your error.

Could you provide some more details? What Linux distribution do you use? 
Did you get the latest libm2k? (identified by commit number 1d1b22917eb8b252468c3544d52abe2762d52ce7 )
Did you install the latest libiio? 
Can you provide a screenshot or copy paste the exact error?

Thank you, 

-Alexandra

mihaiolteanu, 2019-10-03:
@xwkang2019 , any updates?

AlexandraTrifan, 2020-01-06:
Will close this bug. Can be reopened if this still happens.

closed 2020-01-06 by AlexandraTrifan