analogdevicesinc/libm2k #33: Calling context methods on an m2k object fails with "segmentation fault"

mihaiolteanu, 2019-10-04:
This returns the actual device description,
```
Context *context = ContextBuilder::deviceOpen("usb:1.2.5");
string desc = context->getContextDescription();
```
But this fails with a `segmentation fault` error, using the same `context` object from above,
```
M2k *m2k = context->toM2k();
string desc = m2k->getContextDescription();
```
The same thing happens to the rest of the m2k methods that I've tested, like `getAllDevices`, `getSerialNumber`, `calibrateADC`, etc. The same thing happens when the object is instantiated directly with `M2k *context = m2kOpen();`.

M2k extends the Context class and the Context class has some of these methods as public, so it would be expected that these methods can also be called from an M2k instance without any problem.

AlexandraTrifan, 2019-10-04:
I could not reproduce this. If there is another usb context open for the same device, "m2kOpen()" or "deviceOpen()" will return nullptr because it can't create a new context.

If we are trying to call "m2kOpen" or "toM2k" on any generic IIO product except ADALM2000, it will return nullptr.

Indeed, these examples are missing something like: if (context) { do stuff }. The methods that open m2k/pluto/any product, don't throw an exception, they just return nullptr. 
I will add this to the examples and documentation as soon as possible.

AlexandraTrifan, 2019-10-15:
I added some checks for all the available examples in commit 8ab697b94ff298c035fdea572232cad4b04f579b .

closed 2019-10-28 by AlexandraTrifan