analogdevicesinc/libm2k #34: Example binaries fail with "segmentation fault"

mihaiolteanu, 2019-10-04:
[image]

Same for the `analog` and `digital` binaries.
Might be related to [#33](https://github.com/analogdevicesinc/libm2k/issues/33), as this, taken directly from the `examples/analog/getSamples.cpp` file, also fails with "segmentation fault",
```
int main(int argc, char* argv[]) {
    M2k *ctx = m2kOpen();
    ctx->calibrateADC();
    ctx->calibrateDAC();
}

AlexandraTrifan, 2019-10-04:
Similar to #33 . I will return with some commits adding the missing checks in examples and in the libm2k documentation.

AlexandraTrifan, 2019-10-15:
I added some checks for all the available examples in commit 8ab697b .
Also, getSamples was removed from the examples since it was only for testing purposes.

closed 2019-10-28 by AlexandraTrifan