analogdevicesinc/libm2k #48: calibrateDAC causes the waveform generation to fail.

rinatzakirov, 2019-10-20:
Running with ADALM2000, on Windows 10 64-bit, either with Python analog.py example or C analog.exe, analog output doesn't work at all, the waveform never appears on the relevant pins.
For reference, signal generation works fine with Scopy and Alice M2K, just not libm2k.
Also, after running either analog.exe or analog.py first, analog output stops working completely, either through Scopy or Alice M2K. It appears that libm2k crashes something inside the ADALM2000 itself, and it needs to be power cycled to start working in Scopy again.

Commenting out line that calls calibrateDAC() fixes the problem completely.

adisuciu, 2019-10-21:
Hello rinatzakirov,

That is peculiar ... What libm2k version are you using ? Did you connect the W1 output to the 1+ input of the M2K ? Can you try commenting out the triggering part and rerunning the example ? Also after creating the context, Can you run ctx.init() like so ?
```

# This example assumes the following connections:
# W1 -> 1+
# W2 -> 2+
# GND -> 1-
# GND -> 2-
#
# The application will generate a sine and triangular wave on W1 and W2. The signal is fed back into the analog input
# and the voltage values are displayed on the screen

import libm2k
import matplotlib.pyplot as plt
import time
import numpy as np

ctx=libm2k.m2kOpen()
if ctx is None:
	print("Connection Error: No ADALM2000 device available/connected to your PC.")
	exit(1)
ctx.init()
ctx.calibrateADC()
ctx.calibrateDAC()

ain=ctx.getAnalogIn()
aout=ctx.getAnalogOut()
trig=ain.getTrigger()

ain.enableChannel(0,True)
ain.enableChannel(1,True)
ain.setSampleRate(100000)
ain.setRange(0,-10,10)

### uncomment the following block to enable triggering
#trig.setAnalogSource(0) # Channel 0 as source
#trig.setAnalogCondition(0,libm2k.RISING_EDGE)
#trig.setAnalogLevel(0,0.5)  # Set trigger level at 0.5
#trig.setAnalogDelay(0) # Trigger is centered
#trig.setAnalogMode(1, libm2k.ANALOG)

```

-Adrian

rinatzakirov, 2019-10-22:
That is peculiar ... What libm2k version are you using ? 
Tried a few of them, 1) A few days old compiled directly from source code in Visual Studio 2) Pre-compiled libm2k artifact 48644a69 3) Pre-compiled archive libm2k-win64.zip from a couple of days ago, that archive doesn't have a filename that tracks the version, so don't know exactly.
They all behave the same. Tried using Python 2.7 bindings(compiled from source code), Python 3.7(one of the pre-compiled versions), and C code(both the default with trigger that outputs nothing, and custom modded compiled without trigger readings close to zero).

Did you connect the W1 output to the 1+ input of the M2K ? 
Yes I did.

Can you try commenting out the triggering part and rerunning the example ? 
With trigger I don't see anything printed out as expected, since there is no signal to trigger from.
Commenting the trigger out I get a reading average of 0.05 on one channel, and 0 on the other channel, no actual DAC signal present. I also checked with external oscilloscope,there is no output, it is silent, with some minor noise on the line.

Also after creating the context, Can you run ctx.init() like so ?
Tried the code you provided, same exact thing:
* I see the graph with both channels close to 0, one is average of 0.05-0.06, and the other close to zero.
* After closing the window a few times and seeing different values hovering around zero, it finally closes with an error  'module' object has no attribute 'contextClose'

AlexandraTrifan, 2020-01-17:
Hi @rinatzakirov ,

Does this issue still appear on the latest master?

Thank you, 
-Alexandra

rinatzakirov, 2020-01-19:
I just tried it on latest artifact binary caf024d1, and it works!
Thank you.

closed 2020-01-20 by AlexandraTrifan