analogdevicesinc/libm2k #72: Some atributes have changed / nolonger work in libm2k master.402 build?

damercer, 2019-11-29:
Hi:
I just updated my version of libm2k (with Python bindings) and something has changed that renders this error:

Traceback (most recent call last):
  File "C:\ALICEm2k\alice-desktop-2.1.pyw", line 17613, in <module>
    ConnectDevice() # find attached M2K
  File "C:\ALICEm2k\alice-desktop-2.1.pyw", line 8162, in ConnectDevice
    trig.setAnalogCondition(0,libm2k.RISING_EDGE) # RISING_EDGE)
AttributeError: module 'libm2k' has no attribute 'RISING_EDGE'

The example code here seems to still have this usage:

#uncomment the following block to enable triggering
#trig.setAnalogSource(0) # Channel 0 as source
#trig.setAnalogCondition(0,libm2k.RISING_EDGE)
#trig.setAnalogLevel(0,0.5)  # Set trigger level at 0.5
#trig.setAnalogDelay(0) # Trigger is centered
#trig.setAnalogMode(1, libm2k.ANALOG)

Can't find where in the source code the allowed attributes are enumerated and the auto generated help does not list allowed values for the various functions:

https://analogdevicesinc.github.io/libm2k/group__m2ktrigger.html#ga276a274638000400f8c6ccf2efcb18c5

This is another place where things no longer seem to work for aout.push(buffer)

Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Python37\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\ALICEm2k\alice-desktop-2.1.pyw", line 14046, in onAWGBscroll
    UpdateAWGB()
  File "C:\ALICEm2k\alice-desktop-2.1.pyw", line 10153, in UpdateAWGB
    aout.push(buffer)
  File "C:\Python37\lib\site-packages\libm2k.py", line 1582, in push
    return _libm2k.M2kAnalogOut_push(self, *args)
ValueError: Module libm2k error: ERR: Invalid argument - Buffer: Can't create the TX buffer

So what is a programmer supposed to do?

Thanks

Doug

AlexandraTrifan, 2019-12-02:
Hi,

Regarding the first issue, the fix for that example is in this Pull Request. It will be merged today. ( https://github.com/analogdevicesinc/libm2k/pull/67/files ) . 
The attributes are enumerated here ( https://analogdevicesinc.github.io/libm2k/enums_8hpp.html#a6ef409f0fb157999c4050789d0690032 ).

For the second issue: Did you enable any DAC channel before pushing? The error can happen if no channel was previously enabled.

Thank you,
-Alexandra

damercer, 2019-12-02:
Hi:

Yes I think I do. These commands are done as part of the Connectdevice operation on start-up.

    aout.setSampleRate(0, AWGASampleRate)
    aout.setSampleRate(1, AWGBSampleRate)
    aout.enableChannel(0, True)
    aout.enableChannel(1, True)

This was all working a couple of revisions ago.

Also I seem to now be having trouble changing the AWG sample rate and pushing new buffer data on the fly after doing something like this:

    buffer = [AWGAwaveform, AWGBwaveform]
    aout.setCyclic(True)
    aout.push(buffer)

Was working before but now only one of the AWG channels get the new data and makes an output and the other just sits at a random DC level.

What do I need to do differently now?

Thanks

Doug

AlexandraTrifan, 2019-12-03:
Hi,

The build you are using ( https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/29205928 ) is under development (on branch kernel_buffers_analog_out) . We are working on using kernel buffers and streaming on the AnalogOut. 

Can you try the master branch and tell us if you have the same issue?  ( this is the latest master build https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/29186419 )
Could you tell us what sample rates are set for the DAC channels? We will check the bug on the development branch and fix it. 

Thank you,
-Alexandra

damercer, 2019-12-03:
Hi:
The master.396 version seems to have returned everything back to the way it was before and now works as expected. Having master in the name like master.402 for a development branch build is very confusing.

The problem with the .402 build where only one of the AWG channels get the new data and makes an output and the other just sits at a random DC level was happening at all dac sample rates as far as I could tell. Which AWG channel sits at the DC level depended on the order in which the channel waveform data was changed. i.e. if AWG 1 was changed AWG 2 went to DC and if AWG2 was changed AWG1 when to DC. By DC I mean it might have been disabled could not really tell with a simple loop back connection.

Doug

teoperisanu, 2019-12-06:
Hi,

We have worked to fix the issue about pushing samples using cyclic buffer. Can you try master.412 and tell us if it works as you would expect?

Regards,
Teo

damercer, 2019-12-06:
There seems to be something wrong with Appveyor right now, can't down load any of the files. Will try again latter.

teoperisanu, 2019-12-06:
Hi,

I tried to download the files and it seems that Appveyor is back to normal. Can you try again using this link: https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/29336073/artifacts

Thank you,
Teo

damercer, 2019-12-06:
Sorry still not working for me.

[External]

adisuciu, 2020-07-08:
Can I close this or is this still an issue ? 
-Adrian

closed 2020-07-08 by adisuciu