analogdevicesinc/libm2k #77: getSamplesRawInterleaved and manually converting from raw sample to voltage

xwkang2019, 2019-12-09:
Dear Developers,

Thank you for the latest fix on the python bindings which made it possible for us to get samples faster by using the getSamplesRawInterleaved function. However, after playing with the function for a while and looking through the newly implemented convertRawToVolts() and convRawToVolts() functions, I have a few questions regarding the raw samples and its conversion to the float point voltage:

1. The first thing I noticed is that if I call getSamplesRawInterleaved(10000), for example, it will return an array of 40000 samples of integer type. I initially expected to see 20000 because my understanding was that samples of two channels (analog in 1 and analog in 2) are interleaved and thus generating two times of samples... How should I properly interpret the samples returned by the getSamplesRawInterleaved function? (Is that because we are getting 1+,1- ,2+,2- and thus four times of samples?) Since in our project we are only using one channel (1+,1-), I am thinking about extracting the samples using Python slicing (e.g.: data_demultiplexed1=list(data)[0::4]) but not sure which positions of samples correspond to analog input 1.

2. The second issue is that since we have adopted multiprocessing in our communication system, on the receiver side there are two parallel processes: 1. A process continuously call getSamples and pass the samples to the second process through Python queue. 2. A process does all the processing job such as demodulation of OOK samples.
Now I would like to process the raw samples on the second process, but it doesn't have access to the context and cannot control m2k (I called ctx=libm2k.m2kOpen() in the first "getsamples" process already). Now, in order to process the raw samples and converting them to float point voltage, I have the following idea and could you guys kindly check if it makes sense?

1. Use the first process, which has control over m2k, to obtain hw_gain, correctionGain, filterCompensation, offset.
2. Pass these four parameters to process 2 via queue.
3. Plug into the formula ((sample * 0.78) / ((1 << 11) * 1.3 * hw_gain) *correctionGain * filterCompensation) + offset to obtain the samples.

Since we are really concerned about the processing time, I would like to know if hw_gain, correctionGain, filterCompensation, offset are constantly changing while ADALM is running? If not, once the calibration is done and the ADALM is initiated I can only pass these four parameters through queue once and not worrying about them again...

Finally, could you please confirm what functions to call in order to get the hw_gain, correctionGain, filterCompensation, offset? I kind of know getValueForRange gives hw_gain and getFilterCompensation gives filterCompensation but not sure about the other two..

Thank you very much and we appreciate any insights you guys could provide!

Thanks,
Steven

AlexandraTrifan, 2019-12-09:
Hello,

Please keep in mind that the branch you are using, with the modified Python bindings is still under development/review, so some things might slightly change until it is merged to master.

The getSamplesRawInterleaved(10000) will return 4*nb_samples elements in the Python list. In order to get samples faster, we don't copy all the samples into a list, we just return bytes. One sample on one channel for AnalogIn is represented on 2 bytes. That's why you have 40000 instead of 20000. The samples are indeed interleaved, so you will see one sample from CHANNEL_1, then one sample from CHANNEL_2. The difference is that a sample corresponds to 2 elements in the list. 
Let's take an example:
Run: 
```
samples = getSamplesRawInterleaved(10) 
samples.tolist()
```
will return something like this
`[243, 255, 2, 0, 246, 255, 2, 0, 244, 255, 2, 0, 246, 255, 2, 0, 244, 255, 2, 0, 244, 255, 1, 0, 243, 255, 0, 0, 245, 255, 3, 0, 245, 255, 2, 0, 245, 255, 2, 0]`
and `samples.tobytes()`
will return
`b'\xf3\xff\x02\x00\xf6\xff\x02\x00\xf4\xff\x02\x00\xf6\xff\x02\x00\xf4\xff\x02\x00\xf4\xff\x01\x00\xf3\xff\x00\x00\xf5\xff\x03\x00\xf5\xff\x02\x00\xf5\xff\x02\x00'
`
[243, 255] correspond to the first sample on CHANNEL_1
[2, 0] correspond to the first sample on CHANNEL_2

In order to convert all the samples you can use the following chunk of Python code:
```
import struct
#..enable channels, acquire samples
samples = samples.tobytes()
count = int(len(samples)/2)
value_list = struct.unpack('h'*count, samples)   #'h' is the format for short
```

value_list will be similar to the following: (-13, 2, -10, 2, -12, 2, -10, 2, -12, 2, -12, 1, -13, 0, -11, 3, -11, 2, -11, 2)

Next thing is to convert it into Volts. In Python, for AnalogIn() you have 2 exposed methods: convertRawToVolts(channel, raw) and convertVoltsToRaw(channel, voltage).

Running convertRawToVolts(0, value_list[0]) (in the case we described above) should get you the Volts value that you need.

Regarding your questions about the formula parameters: filterCompensation changes if you modify the sample rate, some of the coefficients change if you recalibrate the board, and one might change if you modify the RANGE of the AnalogIn.

For the multiprocessing part: You can open up a context in the second process (without doing any calibration, or modifying any attributes in it) and just use it to run convertRawToVolts on the samples transfered from the first process.
The first process could run m2kOpen() (connect through USB), then calibrate the board, enable channels, set sample rates and run getSamples, then send those samples to the second process.
The second process could run m2kOpen("ip:192.168.2.1"), get a handler for the AnalogIn and just convert all the samples using convertRawToVolts(channel, raw). (The second process will know what values to use for the conversion coefficients).

We will add these explanations regarding new Python exposed methods to the documentation as soon as possible.
Did I manage to answer your questions? Please let me know if something is not clear.

Thank you,
-Alexandra

AlexandraTrifan, 2020-09-01:
These instructions were added to the libm2k Doxygen Documentation.  (  https://analogdevicesinc.github.io/libm2k/#faq )
I think we can close this.

closed 2020-09-01 by AlexandraTrifan