analogdevicesinc/libm2k #98: Potential bug with libm2k windows installer?

xwkang2019, 2020-01-06:
Dear Developers,

This might not be an urgent issue but as we were formalizing the libm2k & libiio installation instructions for students we noticed one thing about the libm2k windows installer:

The Python version we used is 3.7.4.
The libm2k installer (system-setup.exe) is downloaded from this link (master 570):
https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/29924208/artifacts
The libiio installer is downloaded from:
https://github.com/analogdevicesinc/libiio
by selecting the "Primary Installer Package" for Windows.

Previously, we have always run the libiio installer first and then run the libm2k installer, after which we could run our Python programs without issues.

But today we noticed that, if the libiio is not already installed and we run libm2k installer, it will return 3 checkboxes:
1)Install libm2k Python 3.7 bindings
2)Install libiio
3)Install libm2k tools

However, after the installation is finished, if we ran the Python programs we will get the following error message:
    import _libm2k
ImportError: DLL load failed: The specified module could not be found.

Running the libiio installer and finish the installation will fix the error.

Now, in the second test, we first ran the libiio installer to install libiio, and then run the libm2k installer. In this case, the installer is able to detect the already installed libiio and return the following 3 checkboxes:

1)Install libm2k Python 3.7 bindings
2)Overwrite libiio
3)Install libm2k tools

Finish the installation and we can run programs without any errors.

What might be an issue is that the libm2k installer does not actually install the libiio even if the checkbox option "install libiio" is selected. If we look at the installed programs by going to control panel->Uninstall or change a program. Four components need to be there in order for the programs to run without errors: libm2k, libiio, libm2k Python bindings, libiio Python bindings. Running the libm2k installer alone with "install libiio" checkbox selected will still only install libm2k & libm2k Python bindings.

Thanks!
Steven

AlexandraTrifan, 2020-01-09:
Hi, 

Sorry for the delay. We will look into this as soon as possible. We are working on building and testing the installers for all the supported operating systems.

Thank you,
-Alexandra

AlexandraTrifan, 2020-01-13:
Hi,

I ran the libm2k installer from the link you provided on a clean Windows. (Removed all the libiio and libm2k libraries and python bindings.)
I started the libm2k installer, got the following 3 checkboxes and checked them all:
1)Install libm2k Python 3.7 bindings
2)Install libiio
3)Install libm2k tools

After the install, I started Python and managed to run "import libm2k". 

> Four components need to be there in order for the programs to run without errors: libm2k, libiio, libm2k Python bindings, libiio Python bindings. Running the libm2k installer alone with "install libiio" checkbox selected will still only install libm2k & libm2k Python bindings.

After running this installer, I only got libm2k & libm2k Python in the "Add or remove programs" tab, which is fine. The libm2k installer does not run the libiio installer (that's why it does not appear in the previously mentioned tab). It installs the libraries if those are not already present in the system, or if you need them to be overwritten. 

> However, after the installation is finished, if we ran the Python programs we will get the following error message:
import _libm2k
ImportError: DLL load failed: The specified module could not be found.

The error that you encountered appears if the Python specific files (in this case libm2k.py and _libm2k.pyd) are in the right place (which should be something like C:\Python37\Lib\site-packages) but they can't find the libm2k.dll (which should be in C:\Windows\System32 or in some location accessible and mentioned in the PATH environment variable). 

-Did the libm2k installer display any error during installation?
-Could you confirm that in C:/System32 you can find libm2k.dll and libiio.dll after installing libm2k and checking the "Install libiio" checkbox (without installing it separately).

Thank you,
-Alexandra

closed 2020-10-13 by adisuciu