analogdevicesinc/libm2k #99: Missing _libm2k.pyd building on mac

aerialist, 2020-01-07:
Hi,

I'm trying to use libm2k Python binding on Mac.
Followed "Building on OSX" section at
https://wiki.analog.com/libm2k
then ran 
> python setup.py install

It looks the compilation and installation succeed. But importing libm2k fails.

> >>> import libm2k
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/Users/ssato/anaconda3/envs/iio/lib/python3.6/site-packages/libm2k-0.1.0-py3.6.egg/libm2k.py", line 15, in <module>
> ModuleNotFoundError: No module named '_libm2k'

Checking inside egg (and also build directory), there are

* _libm2k.so
* libm2k.py

files but there is no _libm2k.pyd file.

Attached is the log during build.
[build_log.txt](https://github.com/analogdevicesinc/libm2k/files/4028631/build_log.txt)

I'd appreciate any advice what to try to make it work.

aerialist, 2020-01-07:
Mmm... I just learnt that .pyd is only for WIndows. 

Then why libm2k.py cannot find _libm2k module even though _libm2k.so is in the egg??

AlexandraTrifan, 2020-01-07:
Hi,

I need to setup a machine to test your steps. Could you tell me if you have any other Python version installed on your machine?
Can you check that you have the base library installed somewhere visible from your PATH? (you should have a libm2k.dylib which is the base C++ library). The libm2k.py and _libm2k.so are only additions for the Python bindings. 

I will run some tests and get back to you.

Thank you, 
-Alexandra

aerialist, 2020-01-08:
Thank you for your support, Alexandra!

Actually I cannot find libm2k.dylib anywhere.
There is no mention of it in the [build log](https://github.com/analogdevicesinc/libm2k/files/4028631/build_log.txt), neither.
I can see libm2k-sharp-cxx-wrap.dylib created in the build directory.
It's for c sharp binding? Is there special switch needed to create libm2k.dylib?

On my Mac, I basically use Anaconda's Python with its environment management.
For this experiment, I have created a new conda environment called 'iio'

> conda create -n iio python=3.6
> source activate iio
> pip install numpy, 

Then installed libiio by installer package for Mac from
https://github.com/analogdevicesinc/libiio
I had to also download tar ball to install python binding.

AlexandraTrifan, 2020-01-08:
Hi,

Did you run "sudo make install"? 
The "make" command builds the library, but "make install" installs the library in your system. 
That should install libm2k.dylib in a location visible from your PATH. 

-Alexandra

aerialist, 2020-01-09:
Yes, I did run "sudo make install".

I run again and added to log:
[build_log.txt](https://github.com/analogdevicesinc/libm2k/files/4038670/build_log.txt)
(L.760 and onward)

Again, the only *.dylib file appear is libm2k-sharp-cxx-wrap.dylib ...

AlexandraTrifan, 2020-01-10:
Hi, 

The libm2k-sharp-cxx-wrap is related to the C# bindings, so it's not what we are looking for.

I made some changes on a development branch. Could you try those? 
The branch is : https://github.com/analogdevicesinc/libm2k/tree/packaging .
The master version, the one that you were using, was building an OSX framework, libm2k.framework.  From what I tested there was something wrong with the linking path. 
In this updated version (branch packaging), the library is built as libm2k.dylib by default, not libm2k.framework. 

Please remove everything in your build folder before building the new version.
Also, please remove this folder manually: /Library/Frameworks/libm2k.framework .(you should not need it anymore)

Thank you, 
-Alexandra

aerialist, 2020-01-11:
Thank you for your effort, Alexandra!

I tried the packaging branch and libm2k.dylib seem to be created! But still import libm2k results in "No module named '_libm2k'"...

Here is the log
[libm2k_packaging_build_fail.log](https://github.com/analogdevicesinc/libm2k/files/4049405/libm2k_packaging_build_fail.log)

L2: made sure /Library/Frameworks/libm2k.framework is deleted
L4: activate iio environment where libiio and its Python binding is working
L6: git clone libm2k
L15: switch to the packaging branch
L20: cmake
L56: make
L947: sudo make install
L1034: sudo python setup.py install
L1070: /usr/local/lib/libm2k.dylib exist!
L1075: running python in the 'build' folder and import libm2k gets "Segmentation fault: 11"
L1082: running python in other directory and import libm2k gets "No module named '_libm2k'"

rgetz, 2020-01-12:
@tfcollins any thoughts - there was a lot of rpath manipulation we did for some of the Mac frameworks?

aerialist, 2020-01-18:
Thanks again, Alexandra.

fix-rpath modifications resolved ModuleNotFoundError !
But then I'm greeted with Segmentation fault: 11 ...

Running code inside libm2k.py manually line by line, it seems the segmentation fault is raised in "import _libm2k"

[libm2k_fix-rpath_build.log](https://github.com/analogdevicesinc/libm2k/files/4080072/libm2k_fix-rpath_build.log)

L2: activate iio environment where libiio and its Python binding is working
L4: git clone libm2k
L13: switch to the fix-rpath branch
L18: cmake -DENABLE_CSHARP=OFF ..
L65: make
L952: sudo make install
L1037: sudo python setup.py install_lib
L1045: sudo python setup.py install
L1083: running python in the 'build' folder and import libm2k gets "Segmentation fault: 11" <-- this is same behavior as before
L1090: running python in other directory and import libm2k gets "Segmentation fault: 11" <-- this is improvement. It was "No module named '_libm2k'" before
L1099: tried "python -v -m libm2k"
L1274: importing _libm2k raises "Segmentation fault: 11"

AlexandraTrifan, 2020-01-29:
Hi,

Sorry for the delay. It took a bit of searching but I hope this fixes the issue. 
I didn't run into the "Segmentation fault 11" issue until I updated the conda env (which also updated the python version).

I added another commit on the **fix-rpath** branch which should handle the issue. Can you try again and let us know the outcome?

Thanks,
-Alexandra

shunyas, 2020-01-29:
Thank you very much for your continuous efforts, Alexandra!

Well, fix-rpath branch is missing in master now... could you push it?

> (iio) ShunyanoMacBook-puro-1424:libm2k ssato$ git checkout fix-rpath
> error: pathspec 'fix-rpath' did not match any file(s) known to git.

aerialist, 2020-01-29:
Thank you very much for your continuous efforts, Alexandra!

Well, fix-rpath branch is missing in master now... could you push it?

>     (iio) ShunyanoMacBook-puro-1424:libm2k ssato$ git checkout fix-rpath
>     error: pathspec 'fix-rpath' did not match any file(s) known to git.

AlexandraTrifan, 2020-01-29:
Hi,

It was merged to master and removed. 
All those changes should be on **master** now.

-Alexandra

aerialist, 2020-02-02:
Excellent!
Compilation and installation succeeded. Now "import libm2k" seem to work alright.

I will get Pluto hardware back from my friend next week and start testing!

AlexandraTrifan, 2020-02-04:
Hi,

Happy to know it works fine.

Please be advised that libm2k is mainly for interfacing with ADALM2000 devices. ADALM-Pluto (which is an SDR) can also be controlled using this library, but it provides limited functionality (such as configuring sample rates / enabling devices / simple data capture etc.). 

If you are looking for an easy way to configure Pluto in Python, you can look into the **pyadi-iio Pluto example** ( https://github.com/analogdevicesinc/pyadi-iio/blob/master/examples/pluto.py ) .
If you are looking for a C library, libiio is the way to go. ( https://github.com/analogdevicesinc/libiio )

Thank you, 
-Alexandra

aerialist, 2020-02-05:
I meant ADALM2000 actually.

And now I tested and confirmed that libm2k python examples work alright with ADALM2000 on my Mac. Thank you very much!

closed 2020-02-05 by aerialist