analogdevicesinc/libm2k #139: missing some Cmake vars?

rgetz, 2020-05-01:
Following instructions at:
`https://wiki.analog.com/university/tools/m2k/libm2k/libm2k`

```
rgetz@brain:~/github/libm2k/build$ sudo apt-get install git cmake swig g++ python3-dev python3-setuptools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cmake is already the newest version (3.13.4-1).
g++ is already the newest version (4:8.3.0-1).
python3-setuptools is already the newest version (40.8.0-1).
python3-dev is already the newest version (3.7.3-1).
python3-dev set to manually installed.
swig is already the newest version (3.0.12-2).
git is already the newest version (1:2.20.1-2+deb10u3).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
rgetz@brain:~/github/libm2k/build$ rm * -rf
rgetz@brain:~/github/libm2k/build$ cmake ../ -DENABLE_PYTHON=ON
-- The CXX compiler identification is GNU 8.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.20.1") 
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.12") 
---- Building Python bindings
-- Found Python: /usr/bin/python3.7 (found version "3.7.3") found components:  Interpreter Development 
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  bindings/python/CMakeLists.txt:54 (SWIG_ADD_LIBRARY)
This warning is for project developers.  Use -Wno-dev to suppress it.

---- Building C# bindings
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rgetz/github/libm2k/build
```

looking at the warnings:
```
rgetz@brain:~/github/libm2k/build$ cmake --help-policy CMP0078
CMP0078
-------

Starting with CMake 3.13, ``UseSWIG`` generates now standard target
names. This policy provides compatibility with projects that expect the legacy
behavior.

The ``OLD`` behavior for this policy relies on
``UseSWIG_TARGET_NAME_PREFERENCE`` variable that can be used to specify an
explicit preference.  The value may be one of:

* ``LEGACY``: legacy strategy is applied. Variable
  ``SWIG_MODULE_<name>_REAL_NAME`` must be used to get real target name.
  This is the default if not specified.
* ``STANDARD``: target name matches specified name.

This policy was introduced in CMake version 3.13.  CMake version
3.13.4 warns when the policy is not set and uses ``OLD`` behavior.
Use the ``cmake_policy()`` command to set it to ``OLD`` or ``NEW``
explicitly.

.. note::
  The ``OLD`` behavior of a policy is
  ``deprecated by definition``
  and may be removed in a future version of CMake.

```
```
```

AlexandraTrifan, 2020-05-05:
Should be fixed with PR https://github.com/analogdevicesinc/libm2k/pull/140 .

rgetz, 2020-06-30:
thanks - closing.

closed 2020-06-30 by rgetz