analogdevicesinc/libm2k #160: bindings for LABview?

rgetz, 2020-06-30:
Are there bindings for LabView so I can interact with M2k?

Should be straightforward:
https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z0000019Ls1SAE&l=en-US

but might be in a separate repo - similar to the MATLAB bindings are in:
https://github.com/analogdevicesinc/libm2k-matlab

AlexandraTrifan, 2020-12-22:
Added all the needed code in the official repos:
- Vis -  https://github.com/analogdevicesinc/libm2k-labview 
- libm2k wrapper  -  https://github.com/analogdevicesinc/libm2k/tree/master/bindings/labview 
- auto generated html documentation for Vis - https://analogdevicesinc.github.io/libm2k-labview/index.html 
- wiki documentation - https://wiki.analog.com/university/tools/m2k/libm2k/libm2k#labview_bindings 
- the VI package release: https://github.com/analogdevicesinc/libm2k-labview/releases/tag/v1.0.0.1 
As it is mentioned in the release info, for now, a nightly build will be used for libm2k, until the new official version(v0.4.0) is released.

AlexandraTrifan, 2021-02-01:
Enhancement is done, I think we can close this.

closed 2021-02-01 by AlexandraTrifan