analogdevicesinc/libm2k #161: Analog output has issues for small buffer push()

jaxankey, 2020-07-03:
If I use `push([array1,array2])` the two outputs start at the same time, and can remain sync'd. However, if the length of either array1 or array2 gets small, down to 100 or below on my system, I start to see all kinds of jitter and they lose their relative phase.

Here is a minimal script to reproduce it:
```Python
import libm2k
import pylab
import numpy

# Get the first m2k name
name = libm2k.getAllContexts()[0]

# Get the m2k
m2k = libm2k.m2kOpen(name).toM2k()

# Run the calibration
m2k.calibrate()

# Enable the first output, enable it, and send it some data.
ao = m2k.getAnalogOut()
ao.enableChannel(0, True)
ao.enableChannel(1, True)
ao.setSampleRate(0, 75000000)
ao.setSampleRate(1, 75000000)

# Waveforms: sinusoid and square pulse
N  = 200                        # Number of samples to stream
ns = numpy.linspace(0,N-1,N)    # List of integers
v1 = numpy.sin(2*numpy.pi*ns/N) # Single cycle in N points
v2 = numpy.linspace(1,0,N)      # Trigger edge
v2[0] = 0.5                     # But a little smoother...

# Send both waveforms at once
ao.push([v1,v2]) # Default is cyclic, pulse on the first output

# Enable the first input channel, enable it, and take data
ai = m2k.getAnalogIn()
ai.enableChannel(0, True)
ai.enableChannel(1, True)
ai.setSampleRate(100000000)
ai.setRange(0, libm2k.HIGH_GAIN)
ai.setRange(1, libm2k.HIGH_GAIN)

# Now try triggering
trig = ai.getTrigger()
trig.setAnalogMode(1,1)         # ch2, analog condition
trig.setAnalogSource(1)         # ch2
trig.setAnalogCondition(1,0)    # ch2, rising edge
trig.setAnalogLevel(0,0.5)      # ch2, 0.5V

# Set the timeout (5 sec) to be safe if it doesn't trigger
m2k.setTimeout(3000)

# Go for it 10 times
for n in range(10):
    y1, y2 = ai.getSamples(N*3)
    pylab.plot(y1)
    pylab.plot(y2)
```
Notice the N=200 line. As is, this produces perfect synchronization:
[image]

If I set N=100:
[image]

You can see the jitters more clearly as I toggle the number of samples to each analog out individually:
[image]

Notice especially what the later square pulses do. There are glitches and variable delays.

This feels like a buffering issue? I would think for pushed data sets that fit within the analog output's internal buffer, libm2k should just dump the data into there and let the FPGA do the looping internally? This might be a firmware issue.

I am on Linux Mint 19.3 with libm2k v0.2.1 and libiio v0.21.

Keep up the good work! <-- 100% not sarcastic. This library is REALLY moving us forward.

Cheers,
Jack

adisuciu, 2020-07-08:
Thanks for your feedback .. we're looking into it !

-Adrian

teoperisanu, 2020-12-23:
Hi,

After several investigations we concluded there is a hardware limitation for small cyclic buffers. The recommended solution is to use a buffer of at least 1024 samples. If your signal contains a smaller number of samples, you can repeat  the samples in the buffer until the buffer has the length of at least 1024 and is a multiple of 4.
We added this information in the documentation and also, if you are using libm2k with the logging system enabled, a warning should appear.

Thank you,
Teo Perisanu

PR #209 (2020-12-23): M2kAnalogOut: Add warning for small buffers.
Due to a hardware limitation, in cyclic mode the length of the buffers should be greater than 1024.
Add info in the documentation.
Add warning for the end-user.

Fixes #161
closed 2020-12-23 by AlexandraTrifan