analogdevicesinc/libm2k #165: Request for Ubuntu 20.04 compile (or something like flatpak?)

jaxankey, 2020-07-22:
Hi all,

Would it be possible to package v0.2.1 for Ubuntu 20.04? Or (even better) is it possible to distribute to all linux flavors with something like flatpak?

Cheers!

AlexandraTrifan, 2020-09-01:
Hi,

We added Ubuntu 20.04 builds with the following PR ( https://github.com/analogdevicesinc/libm2k/pull/178 ). 
The artifacts will appear in the next release (which will appear as soon as possible). There will be a .deb package for the base library and a .deb package for Python bindings.
We did not decide on a Flatpak package yet. We are working on creating a generic package for libm2k, but it will not be available in the next release.

Thank you,
-Alexandra

AlexandraTrifan, 2020-09-29:
The artifacts for Ubuntu 20.04 were added in the latest release: https://github.com/analogdevicesinc/libm2k/releases/tag/v0.3.0

closed 2020-10-13 by adisuciu