analogdevicesinc/libm2k #211: csharp libm2kPINVOKE threw an exception

jepoy11, 2021-01-07:
[image]

AlexandraTrifan, 2021-01-07:
Hi,

In the MSVC Build configuration, did you select x64 or x86? 
If "Any CPU" is selected, the default configuration might point to x86 (which will mess up the build if your PC is x64).

-Alexandra

jepoy11, 2021-01-13:
Hi Alexandra,

I tried to change the build config to x64 ans x86. but the same error encountered.
Do I miss something in the process? 

[image]

Here is my project files. I got the DLL from LTPowerAnalyzer App program files.
[image]

AlexandraTrifan, 2021-01-13:
Hi,

Can you try to clean the build folder and then build again with the x64 configuration?
Also, did you try to build it in Release mode, or only in Debug mode?

-Alexandra

jepoy11, 2021-01-15:
Hi Alexandra,

I tried all the combination of release,debug, x86 ,x64 .
I got same error.

AlexandraTrifan, 2021-01-15:
Hi,

Could you tell us the location of the 2 dlls? Are they copied in the build directory?
If only libm2k-sharp.dll is copied right next to you binary, then you need to also copy the libm2k-sharp-cxx-wrap.dll over there.
The problem is as follows:
- When you add libm2k-sharp.dll as a reference to your project, the compiler will automatically copy it to the build directory.
- The problem here is that the second dll does not get copied in the right directory and libm2k-sharp (which depends on it) does not know how to find it.

The solution:
- Copy the libm2k-sharp-cxx-wrap.dll manually in the build directory, next to your binary.
- Copy the libm2k-sharp-cxx-wrap.dll manually in System32, such that anyone who needs it will find it.

We will make the necessary changes to the official installer as soon as possible. After that, there will be no need for manual copying of the dll.

Thank you!

-Alexandra

AlexandraTrifan, 2021-02-01:
A new release was published, containing the fix. Will assume this is a fixed,
A new issue can be opened if there are any other problems.

closed 2021-02-01 by AlexandraTrifan