analogdevicesinc/libm2k #220: can I configure two dac channel output at same time

seuliang, 2021-02-01:
I want ADALM2000 to output voltage through different channel at same time, either from dac or digital io output.
Can I do it? I didn't find library function that actually do that.

help,please!

teoperisanu, 2021-02-01:
Hi,

you can output voltage at the same time with both the analogical and digital interfaces.

For the analogical interface use the push function that takes as parameter a list of lists([<buffer1>, <buffer2>] - https://github.com/analogdevicesinc/libm2k/blob/master/bindings/python/examples/analog.py). Make sure to enable both channels.

For the digital interface use any push function, but the output channels must be set as output and must be enabled (https://github.com/analogdevicesinc/libm2k/blob/master/bindings/python/examples/digital.py). A digital sample is a 16-bit number in which the bit at the position k corresponds to the channel k.

Please tell us if you have any problems.

Thank you,
Teo

seuliang, 2021-02-01:
hi,Teo
thank you for reply. I will try it.

Alferd

teoperisanu, 2021-02-08:
Are there any other issues we could help you with? If not I will close this issue.

closed 2021-02-09 by teoperisanu