analogdevicesinc/libm2k #232: Where does libm2k.dll itself get installed?

Videogamer555, 2021-02-24:
I ran the installer, and at no point during installation did it say what folder it would be placed in, nor did it ask me where I would like it placed. I need help locating the file it just installed, so I can copy it to the folder where the program I'm going to compile in Microsoft Visual C++ is going to be located. The dll file must be in the same folder as the exe file in order for a program that uses it to run. I also need to find the lib file (it would be called libm2k.lib) as this contains all the information that the linker (run after the compiler) will need to know about the DLL file.

AlexandraTrifan, 2021-02-24:
Hi,

libm2k gets installed in the system, so the libm2k.dll can be found in System32. 
The libm2k.lib and the include files should be copied at the following path: C:\Program Files (x86)\Microsoft Visual Studio 15.0\VC 

We will add a .zip with all the build artifacts and add it to the releases page from now on.
Hope this is helpful.

Thank you!
-Alexandra

Videogamer555, 2021-02-27:
> Hi,
>
> libm2k gets installed in the system, so the libm2k.dll can be found in System32.
> The libm2k.lib and the include files should be copied at the following path: C:\Program Files (x86)\Microsoft [...]

Thanks for the info.

closed 2021-07-14 by adisuciu