analogdevicesinc/libm2k #234: Python3.8 bindings question

damercer, 2021-04-26:
Now that libsmu is also available for Python 3.8 I've decided to go back and try installing libm2k on my Windows computer. 
I have a newly installed version of Python 3.8.9 64 bit installed.
I down loaded libm2k released version 0.4.0 today. 
After running the libm2k-0.4.0-Windows-setup.exe setup program with the add Python 3.8 bindings checked, I tried to import libm2k in the Python shell.

This is what it spit back:
Python 3.8.9 (tags/v3.8.9:a743f81, Apr  6 2021, 14:02:34) [MSC v.1928 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license()" for more information.
>>> import libm2k
Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    import libm2k
  File "C:\Program Files\Python38\lib\site-packages\libm2k.py", line 15, in <module>
    import _libm2k
ImportError: DLL load failed while importing _libm2k: The specified module could not be found.
>>> 

Is there another step I missed? It finds the \libm2k.py file but not the compiled DLL. Is something missing from the PATH variable.

C:\>path
PATH=C:\Program Files\Python38\Scripts\;C:\Program Files\Python38\;C:\Python27\;C:\Python27\Scripts\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Calibre2\;C:\Program Files\doxygen\bin;C:\Program Files\Inkscape\bin;C:\Program Files (x86)\Analog Devices\libsmu;C:\Program Files\Analog Devices\libsmu;C:\Users\merce\AppData\Local\Microsoft\WindowsApps;C:\Users\merce\AppData\Local\GitHubDesktop\bin;;C:\Users\merce\AppData\Local\Microsoft\WindowsApps

C:\>

Not an urgent problem. I'm not looking to write any M2k Python right away. Just trying to see if libm2k is working for me now.

Thanks

Doug

damercer, 2021-04-27:
I did a little looking around and found libm2k.dll in the System32 folder. It was dated 2/1/21. In the same folder there is a libiio.dll dated 10/24/20 which I verified as the ones placed there by libm2k-0.4.0-Windows-setup.exe (I looked before and after running the set-up program). Are these the right DLL files for use with 64 bit Python 3.8? System32 is in the PATH variable.

Doug

AlexandraTrifan, 2021-04-27:
Hi,

The path looks good. Also, the libm2k.dll and libiio.dll you found in System32 match the ones I got when testing the libm2k installer on a fresh Python38 install, using the version you mentioned above. 
Can you check if libm2k.py and _libm2k.pyd can be found in the following location: C:\Program Files\Python38\Lib\site-packages ?

Thank you!
-Alexandra

damercer, 2021-04-27:
Yes, both files are there and also dated 2/1/21.
Doug

AlexandraTrifan, 2021-04-27:
- Is there another libm2k version (or libm2k Python files) installed for other Python versions? (I saw a Python27 in the Path)

The following image is a dependency tree of the _libm2k.pyd file, the reason for prompting that issue might be that a required .dll is missing.
- Could you check if any of the following dependencies is missing from your machine? Some of them are assumed to be available by default on Windows machines.
[image]

-Alexandra

AlexandraTrifan, 2021-04-27:
We could also try to add C:\Program Files\Python38\Lib\site-packages to the PATH and see if that changes anything.

-Alexandra

damercer, 2021-04-27:
1. There was some old libm2k stuff in the Python27 site-packages from a couple of years ago that I had added by hand before there was an installer. I deleted it and it made no difference.
2. Everything on the dependencies list is there except MSVCP140.dll is lower case ie msvcp140.dll and the same for VCRUNTIME140.dll vs vcruntime140.dll if that matters.
3. I added C:\Program Files\Python38\Lib\site-packages to the system PATH variable and that made no difference.
Doug

AlexandraTrifan, 2021-04-28:
The VCRUNTIME vs vcruntime should not make a difference in this case.
Could you send me the libm2k-wininst.log (should be located in the Python38 directory)?
There are a few other things we can try to find out where this is coming from.
- Can you copy the _libm2k.pyd inside Python38/DLLs directory and try again?
- Can you tell me if the working directory, the location you are running the Python shell contains anything related to libm2k or libiio?
- I will attach a zip containing the release files for libm2k-v0.4.0. This contains all the DLLs and the Python related files.
Can you run the following in a Python shell, opened inside the unzipped directory that I am attaching?
 

`import sys`
`import os`
`for line in sys.path: print(line)`
`for path in os.environ["PATH"].split(";"): print(path)`
`import libm2k`
`print(libm2k.__file__)`

The first 'for' should output the search paths for Python modules: similar to the following:
C:\Python38\python38.zip
C:\Python38\DLLs
C:\Python38\lib
C:\Python38
C:\Python38\lib\site-packages

The second 'for' should output your Path, in the same order as mentioned in the comments above.

The last print, if the import works, should output the location where your Python shell was opened. Because all the required files are contained in the attached zip, running a shell inside that directory, should automatically look for required files/modules in that same location. If not found, then move to the next locations mentioned by the Path.

-Alexandra
[libm2k-win64.zip](https://github.com/analogdevicesinc/libm2k/files/6392448/libm2k-win64.zip)

damercer, 2021-04-29:
1. copying the _libm2k.pyd inside Python38/DLLs directory didn't make a difference.
2. There is nothing (python related) in the working directory except the python source I'm editing
3. Here is the results of running the test script in the extracted folder:
Python 3.8.9 (tags/v3.8.9:a743f81, Apr  6 2021, 14:02:34) [MSC v.1928 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license()" for more information.
>>> 
============ RESTART: C:/ALICEm2k/libm2k-win64/libm2k-win64/test.py ============
C:/ALICEm2k/libm2k-win64/libm2k-win64
C:\ALICEm2k\libm2k-win64\libm2k-win64
C:\Program Files\Python38\python38.zip
C:\Program Files\Python38\DLLs
C:\Program Files\Python38\lib
C:\Program Files\Python38
C:\Users\merce\AppData\Roaming\Python\Python38\site-packages
C:\Program Files\Python38\lib\site-packages
C:\Program Files\Python38\Scripts\
C:\Program Files\Python38\
C:\Program Files\Python38\Lib\site-packages
C:\Python27\
C:\Python27\Scripts\
C:\WINDOWS\system32
C:\WINDOWS
C:\WINDOWS\System32\Wbem
C:\WINDOWS\System32\WindowsPowerShell\v1.0\
C:\WINDOWS\System32\OpenSSH\
C:\Program Files\Calibre2\
C:\Program Files\doxygen\bin
C:\Program Files\Inkscape\bin
C:\Program Files (x86)\Analog Devices\libsmu
C:\Program Files\Analog Devices\libsmu
C:\Users\merce\AppData\Local\Microsoft\WindowsApps
C:\Users\merce\AppData\Local\GitHubDesktop\bin

C:\Users\merce\AppData\Local\Microsoft\WindowsApps
C:/ALICEm2k/libm2k-win64/libm2k-win64\libm2k.py
>>> 
It seems to have found libm2k properly in there.

One thing I noticed that looks funny is these two lines to different site packages:
C:\Users\merce\AppData\Roaming\Python\Python38\site-packages
C:\Program Files\Python38\lib\site-packages
The frist one might be a leftover from an earlier failed attempt at installing Python3.8.9? I had uninstalled it before redoing the current one to a different place that worked.
Thanks
Doug

AlexandraTrifan, 2021-04-29:
Hi,

The path you mentioned is the only thing that does not look right here. I think it might be a leftover if you tried to install Python just for the current user (i am not 100% sure).
Is there anything that might disturb the libm2k.py files at that location?
Since that path has priority over the correct site-packages path, it might lead libm2k.py to a wrong location for _libm2k.pyd.
Can you try to remove it from PYTHONPATH?
You can either run 
`import sys`
`sys.path.remove('C:\Users\merce\AppData\Roaming\Python\Python38\site-packages') `

But I think that will only be applied until you close the Python shell.
Or you can go to the ENV variables in Windows and check whether that faulty path is listed under PYTHONPATH.

Thank you!
-Alexandra

damercer, 2021-04-29:
I looked in that Roaming folder and it only had numpy in it ( I guess that is where PIP3 put it). I moved the numpy folders to the site_packages folder under Python38 and deleted the Roaming stuff. numpy still loads now. That line is missing from the Path but still can't import libm2k (except in that test directory).

AlexandraTrifan, 2021-04-29:
Do you have a 64bit or 32bit machine?

damercer, 2021-04-29:
Quad core I5 64 bit processor running Win 10 2004 64 bit OS

damercer, 2021-05-01:
I think I have identified what is (was) missing on my system.
I ran the m2kcli tool in a command window and it said that msvcr120.dll was missing.
I copied the msvcr120.dll file you included in that zip file into the System32 folder (where all the other .dll files are) and I can now import libm2k in the Python38 shell.
Here is a snapshot of what shows up in the program manager as for Microsoft Visual C++:
[image]

One of these must not include the right runtime to match what you are using on your development machines for libm2k.

Your libm2k installer needs another check to see if this dll also needs to be installed on the end user's system.

Thanks

Doug

AlexandraTrifan, 2021-05-04:
Hi,

Thank you for looking into this. I will add the missing DLLs in the building/packaging process and commit the changes to master.

-Alexandra

closed 2021-05-04 by damercer

mujum96, 2022-01-06:
> I think you have not added this dll, I tried with the latest version and it only worked after copying that dll.

damercer, 2022-01-09:
Adding a comment to a closed issue probably will not get the developers attention. If you still have an issue on this topic re-open the issue or make a new open issue.

mujum96, 2022-01-11:
@damercer I could sort this issue out by the same DLL method that you have mentioned earlier. But a better help was the already closed issue #98. Thank you for your comment though, I will keep this in my mind.