analogdevicesinc/libm2k #235: LabVIEW NXG support

nelepost-jpg, 2021-05-15:
Is it possible to add NI LabVIEW NXG support? Current version doesn't support NXG branch thus I am unable to install library using NI package manager (.gvi for VI's and .gll for library package).

rgetz, 2021-05-15:
Did you check out https://github.com/analogdevicesinc/libm2k-labview

- Robin

nelepost-jpg, 2021-05-16:
@rgetz Thanks for quick reply.

Yes, I do

I found .vip file to install (https://github.com/analogdevicesinc/libm2k-labview/releases/tag/v1.0.0.1), but Labview NXG doesn`t support this file type. It uses .nipkg (https://www.ni.com/documentation/en/labview/latest/application-builder/types-of-distributions/) for library intalation.

I also tryed to make project code conversation by NI Code Conversation Utility for Labview NXG 5.0. When I am opening converted examples i get error massage "Conversion was unsuccessful because the file is from an unsupported version of LabVIEW. This file was last saved in LabVIEW 2015, and the minimum supported version is LabVIEW 2017. Save the file in a supported version of LabVIEW and reconvert". 
[image]

Its possible to re-save file in LabVIEW 2020 and re-convert, but this method wont add instrument pallete. Its also possible for me to create .nipkg from converted file by my own and share via github if you interested in.

AlexandraTrifan, 2021-05-17:
Hi,

The .vi files from the official repository were saved last in the 2015 format, to allow any version of LabVIEW to open them. We used those .vis on the libm2k-labview repository in order to make the interaction easier for any version people are using.
We focused on getting the VIs working on all the LabVIEW versions and the .vipkg provided in the releases page should work for both LabVIEW 2015 (or any older version) and 2020. We did not focus on NXG as it seems that NI decided not to support and develop it further. 
https://forums.ni.com/t5/LabVIEW/Our-Commitment-to-LabVIEW-as-we-Expand-our-Software-Portfolio/td-p/4101878?profile.language=en

If you have the LabVIEW 2020 installed, you can clone the libm2k-labview official repository and open the ADALM2000.lvproj file. Then, as you mentioned, every file can be re-saved into the 2020 format, including the .ctl files, which are part of the instrument pallete. If that is not possible, we can push a new branch in the libm2k-labview repo with the 2020 re-saved files. 
If you manage to create the .nipkg file and you wish to share it, you can do so on the libm2k-labview repository. 

Please let us know if you have any updates or issues going further. 

Thank you!
-Alexandra

nelepost-jpg, 2021-05-18:
Ok, I will share .nipkg when its done. I need some time to double check if it works correctly on NXG version.

closed 2021-05-18 by nelepost-jpg