analogdevicesinc/libm2k #250: SPI API chip_select output

xwkang2019, 2021-07-01:
Dear developers,

We saw Adrian's slides for GRcon19 and were trying to tune HMC833 PLL by writing to its register via the SPI interface. We were able to have the desired clock and data signal by calling spi_write_and_read and using the SPI_MODE parameter, but the chip_select parameter seems to be working in the opposite way of what we wanted. For HMC833, we want an enable signal that goes to HIGH during a SPI frame and stays low otherwise. It seems that if I set spi_init_param.chip_select = 0, DIO 0 will stay high and goes low during a SPI frame...

Any suggestions are appreciated!
[image]

teoperisanu, 2021-07-02:
Hi,

chip_select is the parameter that stores the index of the digital channel on which the chip select event will occur. The scenario you are describing (chip select event on high) is not currently support by our API, the event triggers only when the CS goes low.
We will work on this modification in order to allow the user to control the behavior of the CS and we will get back to you as soon as everything is done.

Thank you,
Teo

teoperisanu, 2021-07-02:
I added the possibility of changing the CS polarity in #251. In order to change the polarity of the CS set the cs_polarity to ACTIVE_HIGH (https://github.com/analogdevicesinc/libm2k/blob/modify_spi_cs_behavior/bindings/python/examples/spi.py#L52). Can you confirm that the result is the one you expected?

xwkang2019, 2021-07-02:
Thanks for the enhancement! We are using the python bindings by installing libm2k through the Windows installer via https://github.com/analogdevicesinc/libm2k/releases. What is the best way we can test these new changes? I remembered in the past for each new commits there is an updated windows executable installer uploaded in Appveyor-Artifacts but now we don't have them anymore... (e.g., as in https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/38731656/artifacts).

teoperisanu, 2021-07-05:
The CI is down for the moment. I will get back to you with the artifacts when it is fixed. Meanwhile, if you want you can try to build it for Windows following the instructions from the wiki (https://wiki.analog.com/university/tools/m2k/libm2k/libm2k#building_on_windows).

xwkang2019, 2021-07-07:
Thanks. A windows installer for the Python bindings will be much appreciated. I did try build it myself following the windows build instructions and was able to generate the visual studio solution without error using cmake-gui. I then built the solution on VS and obtained files such as setup.py, libm2k.dll, libm2k.py and _libm2k.pyd. Finally, after copying the dll files to System32 and sysWOW64, I ran the command "python setup.py bdist_wheel" in cmd to install the python bindings. 

However, I was not able to import libm2k successfully, the error message says:
[image]

Not sure what's causing the problem..

xwkang2019, 2021-07-07:
My system is 64 bit, so is my Python. I also checked x64 as the "optional platform for generator" and set IIO_LIBRARIES to libiio/MS64/libiio.lib as the instructions suggest.

AlexandraTrifan, 2021-07-08:
Hi,

So you used the .whl artifact built by "python setup.py bdist_wheel" and installed it in the system, then got the error posted above?
This means that some of the libs/packages were build for 32bit instead of 64bit.

In Visual Studio, while building, did you choose "Release" and "64" for the C++ compiler configuration? (example attached below)

[image]

-Alexandra

xwkang2019, 2021-07-09:
Thanks for the quick guidance! I double-checked that I have set the building to "Release" and x64 during the VS configuration but was still getting the same error after attempting to build. I have shared my build folder on the following link (including all the dependencies downloaded and used for building libm2k) in case that could help us locate where the problem is.
https://drive.google.com/file/d/1XpqIkY9KMaNq10lw4lADqIPC5CFnFXHQ/view?usp=sharing 

I followed the instructions on https://github.com/analogdevicesinc/libm2k/wiki/Building-libm2k-on-Windows step by step.
I previously had libm2k and libiio installed through exe installer and working in my system. I uninstalled them using "Add or remove program" of windows before rebuilding using CMake, Swig and VS.

The libm2k folder I used for building is fetched from the latest branch Teo created to add the new SPI feature.

Thanks ahead for any further instructions!

AlexandraTrifan, 2021-07-12:
Hi,

I used the build folder you provided, ran a Python 3.7 terminal in the build folder and it seems that your build works for me. I also copied all the .dlls in the System32 folder and cleaned up other versions of libm2k to be sure.
Also, I installed the .whl you created using "python3 -m pip install libm2k..-0.4.0.whl" and that one seemed to work as well.
Could you please run a Python terminal inside the "build" folder, run the following commands and send me the output?
"import _libm2k"
"print(_libm2k.__file__)"

Also, in the python terminal run the following to check the path:
"import sys
for path in sys.path:
	print (path)"
	
-Alexandra

xwkang2019, 2021-07-12:
Thanks for the follow-up!

Below is the terminal output for printed path:
C:\Users\ND Wireless\AppData\Local\Programs\Python\Python37\python37.zip
C:\Users\ND Wireless\AppData\Local\Programs\Python\Python37\DLLs
C:\Users\ND Wireless\AppData\Local\Programs\Python\Python37\lib
C:\Users\ND Wireless\AppData\Local\Programs\Python\Python37
C:\Users\ND Wireless\AppData\Local\Programs\Python\Python37\lib\site-packages
C:\Users\ND Wireless\AppData\Local\Programs\Python\Python37\lib\site-packages\win32
C:\Users\ND Wireless\AppData\Local\Programs\Python\Python37\lib\site-packages\win32\lib
C:\Users\ND Wireless\AppData\Local\Programs\Python\Python37\lib\site-packages\Pythonwin

I am getting the previous error if I try import _libm2k
>>> import _libm2k
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: DLL load failed: %1 is not a valid Win32 application.

I then tried copy all the .dll files in libiio to System32 and also copied libm2k.dll file to System32. Then I built and installed the wheels as follows:

C:\Users\ND Wireless\Documents\buildm2k\build\dist>python -m pip install libm2k-0.4.0-py3-none-any.whl
Processing c:\users\nd wireless\documents\buildm2k\build\dist\libm2k-0.4.0-py3-none-any.whl
Installing collected packages: libm2k
Successfully installed libm2k-0.4.0
You are using pip version 19.0.3, however version 21.1.3 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

C:\Users\ND Wireless\Documents\buildm2k\build\dist>python
Python 3.7.4 (tags/v3.7.4:e09359112e, Jul  8 2019, 20:34:20) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import libm2k
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\ND Wireless\AppData\Local\Programs\Python\Python37\lib\site-packages\libm2k.py", line 15, in <module>
    import _libm2k
ImportError: DLL load failed: %1 is not a valid Win32 application.
>>>

Is this happening because I did not entirely clean the previous libiio and libm2k (I only uninstalled them using "Add or remove programs")? If so how could I completely get rid of previous versions of libiio & libm2k installed?

Thanks!

AlexandraTrifan, 2021-07-13:
Hi,

Could you check the location of the following files:
- libm2k.py -> should be in <Path to your Python location>\Python37\Lib\site-packages\libm2k.py
- _libm2k.pyd -> should be in the same location as libm2k.py
If you find those files in any other locations in the Python37 folder, you should remove them.

Could you also check what files are under the following path: C:\Users\ND Wireless\AppData\Local\Programs\Python\Python37\lib\site-packages\win32 ? (I am asking because I haven't seen this path before and it looks a bit weird)

I can't say for sure if something was not completely cleaned up after uninstalling. "Add or remove programs" calls the "uninstall" script for each of these libraries, which should get rid of the base libraries. But you should also see something like "Python 3.7 libm2k v0.4.0" in the "Add or remove programs" menu. If you find one entry like that, you should also uninstall it. 

-Alexandra

xwkang2019, 2021-07-13:
Hi, I searched both libm2k.py and _libm2k.pyd under Python37 and there is only libmk2.py in the current address (\site-packages). There are two _libm2k.pyd, one under \site-packages and another under \DLLs
[image]

Below are the list of files inside the win32 folder:
[image]

I deleted the _libm2k.pyd file under the \DLLs directory (I have previously already removed Python 3.7 libm2k v0.4.0) and then tried to import libm2k again from the terminal but is still getting the same error:

```
Python 3.7.4 (tags/v3.7.4:e09359112e, Jul  8 2019, 20:34:20) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import libm2k
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\ND Wireless\AppData\Local\Programs\Python\Python37\lib\site-packages\libm2k.py", line 15, in <module>
    import _libm2k
ImportError: DLL load failed: %1 is not a valid Win32 application.

```
Does this win32 site -packages suggest something wrong with my Python? It worked well with the previous libm2k and libiio installed using the installer released.

xwkang2019, 2021-07-13:
I also checked another computer which currently we have the latest released libm2k working (installed using the installer) and the printed path look the same (e.g., has Python37\DLLs, Python37\lib\site-packages\win32)

We installed this version of Python from the link: https://www.python.org/downloads/release/python-374/
and selected Windows x86-64 web-based installer.

AlexandraTrifan, 2021-07-14:
Hi,

Could you clean up all the installed versions and try to use the Windows setup installer? 
I added some fixes to the nightly build and also added the changes Teo mentioned earlier.
You can get it from this development build: https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/39990626/artifacts .

-Alexandra

xwkang2019, 2021-07-14:
The windows installer worked for us and we were able to test the new SPI feature, working as desired.
Thanks a lot for getting the windows build installer released!

adisuciu, 2021-07-14:
Great ! We're preparing a new release in the coming month that will contain these fixes. Stay tuned.

closed 2021-07-14 by adisuciu