analogdevicesinc/libm2k #253: Unable to create smooth analog output in streaming mode

rfchainer, 2021-07-05:
Hello, I have been trying to use libm2k to create a sine wave using my ADALM2000 (of about 7MHz) and am unable to do so in streaming mode. I am feeding a sine wave input and have tried all sorts of combinations of buffer size and number of kernel buffers, but whenever i try to do a streaming push (even with various amounts of delays), the analog output stops until the push is complete, which seems like it is unintentional. I am wondering if there is a way to use multiple buffers so that one buffer can be written while the other is read, and use that for my modulation.

Below are some examples (those are using gr-m2k, but I replicated the exact same results with the C++ libm2k as well, and the same problem occurs there). The spectrum analyzer on the right size is a separate SDR which is connected to the output of the M2K (I did not have another oscilloscope on hand so I am using that for visualization). When cyclic mode is enabled, everything works perfectly and there is a sharp peak at the desired frequency. However, in non-cyclic mode (but with a cyclic input of a simple sine wave), the output pulses on and off repeatedly.

[image]

[image]

Is there any way to create a smooth sine wave using just libm2k and analog output without having to use cyclic? (if I use cyclic then I have the same issue where when I want to change the frequency, the output pauses while the new buffer is pushed, which is not acceptable for my use-case)

teoperisanu, 2021-07-05:
Did you try the example (https://github.com/analogdevicesinc/libm2k/blob/master/examples/analog/streaming_one_channel.cpp). We will investigate (both the gr-m2k and the C++ API) and get back to you as soon as possible.

rfchainer, 2021-07-05:
I see, that makes sense about the USB load issue. Would it work if I were to implement my modulation on the m2k within Linux itself (using libiio) or am I forced to stream data in over USB? 

Also - is there a way to have multiple buffers, each cyclic, and choose between them quickly without having to load data every time (effectively a way to have a slow FSK modulation where I can vary between 2 different frequencies separated by 50Hz or so)?

teoperisanu, 2021-07-05:
1) Yes you are right. You would get rid of this limitation if you used the high speed interface on the local backend (https://wiki.analog.com/resources/tools-software/linux-software/libiio_internals#high-speed_mmap_interface). Also, probably the next m2k firmware will contain libm2k.

2) No, you cannot have more than one buffer for a device.

rfchainer, 2021-07-05:
Alright. Is there any repo or instructions for building libm2k for the ARM Linux on the m2k (ideally just storing it in RAM, without having to rebuild the whole OS image), or would I have to figure out the cross-compile stuff myself?

adisuciu, 2021-07-06:
We don't have instructions specifically for the m2k but there are for the Pluto: https://wiki.analog.com/university/tools/pluto/devs/embedded_code?s[]=cross&s[]=compile

You simply need to get M2k's sysroot as opposed to Pluto's sysroot and build against that. 

The current firmware image only supports libiio. We don't currently distribute sysroot for m2k so you will have to build it yourself when building from the firmware image - https://wiki.analog.com/university/tools/pluto/building_the_image?s[]=build&s[]=firmware. You need to do 
`make sysroot` 
once you set everything up.

We added the commit that does that here - https://github.com/analogdevicesinc/m2k-fw/commit/77bdf2f662f253cb3fb8ba20884b4746dcaa70f2 . The process is straightforward once you install the toolchain - You need to install Xilinx WebPack 2019.R1 as well as some other dependencies (on Linux). 

Starting with the next firmware version we will distribute sysroot for m2k and have libm2k be part of that sysroot so you won't have to compile it yourself.

-Adrian

xwkang2019, 2021-07-22:
We were able to stream over one single channel at output sampling rate of 750 kHz (at least the output looks continuous and there is no visible gap between the buffers we pushed). However, when we tried to write output buffers with the same buffer size but now including both channels of data, we had to introduce time delay (e.g., time.sleep(0.01)) between aout.push() calls to get the output signal. If we reduce the time delay, something seemed to crash and M2K won't generate any output signal.

Is the 750 kHz the maximum sampling rate for streaming over 1 channel only? How will different kernel buffer size affect the streaming performance? Does using APIs such as pushRawSamplesInterleaved() allow us to push buffer faster than push() and therefore have less gap between each buffer pushed?

Thanks for any insights!

teoperisanu, 2021-07-23:
When both DACs are enabled, the push function will try to synchronize them. The synchronization algorithm detects if the DMA is at the first buffer reading and blocks both DACs until enough data (one buffer for each DAC) arrives.
In the case you are describing in which the m2k crashes, probably the DACs are blocked and waiting for data. An important remark is that at 1 kernel buffer the push function on both channels won't work. Also the push function is blocked when all kernel buffers are filled with data and the flow will be resumed only when one buffer is consumed.
A suggestion is to see what is the value of dma_sync (found in the m2k-dac-a or -b device) when your program blocks. If dma_sync = 1 then your output is blocked until someone turns if off (normally libm2k should do that for you, if not this can be a bug). Also a code snap can be really useful in order to detect the problem.
From our tests 750 kHz seems to be the maximum samplerate for streaming data. The number of kernel buffers (max 64) allows you to queue buffers. This can be very useful in some cases but also can affect your memory. The kernel buffers gives you more time to spend between pushes, if enough kernel buffers are filled with data. Streaming data requires at least 2 kernel buffers (default is 4).
The function pushRawSamplesInterleaved will spare you some time, because there will be no conversion from volts to raw (the data computation will be left to the user). In some cases this function can help with the gaps, but at a low samplerare, even the normal push function will work. The idea is to fill a new buffer before the current buffer is consumed.
Maybe this example will help you: https://github.com/analogdevicesinc/libm2k/blob/master/examples/analog/streaming_synchronized.cpp
The easiest way to stream data on both devices is to have buffers of same length, the same number of kernel buffers and the same samplerate. These parameters can vary, but some extra computation would be necessary (https://github.com/analogdevicesinc/libm2k/blob/master/examples/analog/sync_stream_diff_frequencies.cpp).

Feel free to ask if something is unclear.
Teo

closed 2024-02-14 by adisuciu