analogdevicesinc/libm2k #256: M2K SPI Interface: Reading MISO input

xwkang2019, 2021-07-13:
Dear Developers,

We were able to use M2K SPI interface to successfully write to the register and tune the HMC833 PLL (for now we are setting V+ to 3.3V and 0V back and forth to emulate the ENABLE signal before we are able to build the latest version of libm2k with the SPI chip-select polarity enhancement).

We now would like to read from the chip but are not sure how to properly read the SPI signal received from the MISO pin as it was not shown in the example Python script in the libm2k repo: 

`m2k_spi_init.miso = 7  # dummy value - miso is not used in this example`

Thanks in advance for any suggestions you could offer!

adisuciu, 2021-07-14:
The dummy value represents the pin that is used for MISO. You simply use the same API spi_write_and_read, and the data is returned in the data parameter. 

Some C++ examples: 
https://github.com/analogdevicesinc/libm2k/blob/master/tools/m2kcli/commands/communication/spi.cpp#L136
https://github.com/analogdevicesinc/libm2k/blob/9576f38420609efb32c9fc050d36f489f98be5cf/tools/communication/examples/spi/main.cpp

-Adrian

xwkang2019, 2021-07-14:
Worked. Thanks for the clarification!

adisuciu, 2021-07-14:
No problem

closed 2021-07-14 by adisuciu