analogdevicesinc/libm2k #277: Dual channel analog capture using External Trigger in TI Pin

codenio, 2021-09-16:
I am trying to use External Trigger Input Pin TI for initiating dual-channel analog capture in ADALM2000 using the python binding.
I am unable to find the exact ways of implementing the same from the document and have the following queries.

1) is this possible..?
2) if yes, is the below code correct

    ```python
   # Get Trigger Object
    trig=ain.getTrigger()
   # Set Analog Trigger Mode as External on TI pin
    trig.setAnalogMode(libm2k.EXTERNAL)
   # set Analog Channel 1  External Trigger Condition as RISING_EDGE_DIGITAL on TI Pin
    trig.setAnalogExternalCondition(1,libm2k.RISING_EDGE_DIGITAL)
   # set Analog Channel 2 External Trigger Condition as RISING_EDGE_DIGITAL on TI Pin
    trig.setAnalogExternalCondition(2,libm2k.RISING_EDGE_DIGITAL)
    ```
3) if not, is there an alternative way for doing the same?

adisuciu, 2021-09-17:
You are missing trig.setAnalogSource(0).
This may be a little confusing, but the trigger interface basically works like this: you have two *TRIGGER* channels - which are different from the regular analog channels. The channel's data is always acquired synchronously, but you can select one of the two *TRIGGER* channels to synchronize on. Therefore the code should go like this:

```
trig=ain.getTrigger()
trig.setAnalogMode(0, libm2k.EXTERNAL) # this will set TRIGGER channel 0 to external
trig.setAnalogExternalCondition(0,libm2k.RISING_EDGE_DIGITAL) # this will set TRIGGER channel 0 
trig.setAnalogSource(0) # set TRIGGER channel 0 as the source for the analog interface

ain.enableChannel(0, True)
ain.enableChannel(1, True)
ain.getSamples(1024) # the data should only be acquired when external trigger is present

```
-Adrian

codenio, 2021-09-17:
Thanks, @adisuciu  for the prompt response, to my surprise using the previously mentioned code (without setAnalogSource()) works as expected. wondering TRIGGER channel 0 is configured as default.

Adding examples and use-cases in the documentation would remove ambiguity with the configuring Triggers 

-Aananth