analogdevicesinc/libm2k #278: DMM units

adisuciu, 2021-09-22:
DMM units are a little messed up. They have spaces, some of them are short form, some of them long form.
I think we should have both short and long form (e.g. short form:"V", long form:"Volts" ; short form: " °C" ; long form "Celsius"; short form: "Pa"; long form "Pascal"

https://github.com/analogdevicesinc/libm2k/blob/master/src/analog/dmm_impl.cpp#L119

rgetz, 2021-09-26:
I think the nomenclature is unit “name” and “symbol” - which is already defined by the International System of Units and is listed here:

https://www.rapidtables.com/electric/Electric_units.html

and

https://www.bipm.org/en/measurement-units

AlexandraTrifan, 2022-09-01:
Included in the latest release.

closed 2022-09-01 by AlexandraTrifan