analogdevicesinc/libm2k #279: Add python 3.9 support

jpnielsen, 2021-09-26:
Please add python 3.9.x support

Python 3.9 was released 2020-10-05

rgetz, 2021-09-27:
Did you try Python 3.9 and run into issues/bugs?

jpnielsen, 2021-09-27:
C'mon - What kind of question is that - If I run python 3.9, should I
choose 3.7 or 3.8 ?????

[image: image.png]

jpnielsen, 2021-09-27:
The installer lets me chose python 3.7 or 3.7 - not 3.9

https://github.com/analogdevicesinc/libm2k/releases/download/v0.5.0/libm2k-0.5.0-Windows-setup.exe

I was hoping you would keep this software updated.
[image]

AlexandraTrifan, 2021-09-27:
Hi, 

I am looking into adding 3.9 to the build. Will come back with updates.

Thank you!
-Alexandra

AlexandraTrifan, 2021-10-20:
Hi,

Python 39 was added to the installer.
This will be included in the next official release, but can be tested using the nightly builds.
The latest nightly build can be found here: https://ci.appveyor.com/project/analogdevicesinc/libm2k/build/artifacts

-Alexandra

AlexandraTrifan, 2021-10-20:
Hi,

It seems that you checked Python 3.10 when the installer required a path to your Python installation.
The bindings were built for Python 3.9. The installer can't check the python version you provided, so you need to make sure you provide the correct version.
These bindings are using a compiled python wrapper/layer which links to the specific Python version they were built with. Using them with another Python version can lead to undefined behaviour, like the one specified above: where the wrapper _libm2k.dll can't be found. 
If you have the Python3.9 installed, you can run the new installer and point it at that specific version.

Thank you!
-Alexandra

jpnielsen, 2021-10-20:
It kind of works when I installed Python 3.9 instead

But it said that my device was not connected.

I fixed that by installing the drivers and iio - apparently this libm2k installer didn't install everything needed to communicate with firmware 0.28
```

https://github.com/analogdevicesinc/plutosdr-m2k-drivers-win/releases/latest
https://github.com/analogdevicesinc/libiio/releases/latest
```