analogdevicesinc/libm2k #282: Using other unused DIO pins after configuring the M2K-SPI interface

xwkang2019, 2021-10-05:
Hello, 

Just a general question regarding the functionality of the Digital I/O. When configuring the M2K-SPI interface (for the record we are using the Python bindings), we designate certain pins to be used for the SPI interface using APIs such as: m2k_spi_init.clock,  m2k_spi_init.mosi and m2k_spi_init.miso. My question is, after configuring the SPI interface, could we still use the rest of the DIOs that are not used for the SPI interface?  

I assume we can still use the other DIOs at least for writing (based on what I see from the example script where the unused pin 3 and pin 4 are used for writing):
https://github.com/analogdevicesinc/libm2k/blob/master/bindings/python/examples/spi.py

But can we read from the other unused pins as well after configuring the M2K-SPI interface? My recollection is that when M2K digitalIn reads, it reads from all 16 pins simultaneously and return a single integer/bit array. Not sure if this will be in conflict with the existing SPI pins that are configured.

Thanks in advance for any information you could provide!

teoperisanu, 2021-10-06:
Hi,

You cannot use concurrently the SPI interface and the Digital interface (with buffer), because the spi_write_and_read uses both the RX and TX buffers. However, you can modify the raw state of a pin while a buffer operation is performed. If you want channels to maintain a state while you are doing buffer operations, simply don't enable them, but set them to output and set their raw state.

There is a prototype for libm2k for combing the SPI and the Digital interface (it is still under development), but if you are interested:
- the code: https://github.com/analogdevicesinc/libm2k/tree/expose_spi_create_buffer 
- the artifacts: https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/40455674/artifacts
The above version introduces two new functions for SPI:
- spi_create_buffer - creates the digital buffer that would normally be created by spi_write_and_read
- spi_write_and_read_samples - initiate the SPI protocol based on a given buffer
The idea behind these functions is that you can create a buffer for SPI, then modify the buffer in order to obtain the desire data for the other channels and then start the SPI data exchange.

Python example:
`# set init parameters
spi_desc = libm2k.spi_init(spi_init_param)
data = bytearray([0x4A, 0xF1])
buffer = libm2k.spi_create_buffer(spi_desc, data) # create TX buffer
# buffer manipulation
libm2k.spi_write_and_read_samples(spi_desc, buffer, data)
# MISO bytes are now in 'data'`

- Teo

vsethia-adi, 2021-12-09:
Do you have any timelines on when to include the functions spi_create_buffer and spi_write_and_read_samples in the main library.
Reason is that I need to modify the non-SPI pins while SPI functionality is in use.

teoperisanu, 2022-01-03:
We don't have a fixed timeline, but the new functionality will be included in the next release, which should come in the following months.
Meanwhile you can use these artifacts: https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/42065600/artifacts

I'll close this issue - the discussion will continue in #297.

closed 2022-01-03 by teoperisanu