analogdevicesinc/libm2k #290: In the Python bindings libm2k.getVersion() returns wrong version

trinamic-bp, 2021-11-12:
Hi,
I just installed the windows Python 3.8 wheel version 0.5.0.
But if I call `libm2k.getVersion()`, `'v0.4.0-g936e2c8'` is returned. 
```
>>> import libm2k
>>> libm2k.getVersion()
'v0.4.0-g936e2c8'
```
Also `pip freeze` says 0.5.0 is installed.

You may single source the package version: [https://packaging.python.org/guides/single-sourcing-package-version/](https://packaging.python.org/guides/single-sourcing-package-version/)

AlexandraTrifan, 2021-11-12:
Hi, 

Did you also install the 0.5.0 libm2k, or just the python wheel?

The libm2k python bindings display the version of libm2k (the main C++ library) The wheel you installed was built against libm2k 0.5.0. But if you did not reinstall the libm2k library, the python bindings may use the version found on your PC. This might be the reason for seeing 0.4.0.
These versions should be compatible because there were no major API changes.

Thank you!
-Alexandra

trinamic-bp, 2021-11-12:
Oh yes, sorry about this one, you are right. I thought the wheel is also a binary extension.

closed 2021-11-12 by trinamic-bp