analogdevicesinc/libm2k #294: spi_write_and_read() func does not work in version 0.5

vsethia-adi, 2021-11-30:
I have installed the latest libm2k 0.5 recently to check for the spi_create_buffer func. Error I get is that this functionality does not exist.
AttributeError: module 'libm2k' has no attribute 'spi_create_buffer'

Furtherthe spi_write_and_read() func stops working when called for the second time. It is always waiting for the data to be read back by ADALM2k.

Error message I got is the following, if I disconnect the ADALM2K USB port to break the deadlock.
ERROR: READ LINE: -32
ERROR: READ INTEGER: -32

Please let me know how to get the spi_write_and_read func working again.

AlexandraTrifan, 2021-12-02:
Hi,

Could you provide the example you are using to make libm2k v0.5.0 crash when calling spi_write_and_read(), in contrast with libm2k v0.4.0? 
Also, could you try to uninstall libm2k and the Python bindings before installing another version and check whether this fixes anything?

The "spi_create_buffer" is not included in the latest development build or the libm2k v0.5.0. It was mentioned in a discussion : https://github.com/analogdevicesinc/libm2k/issues/282 , but it is still a prototype under development.

Thank you!
-Alexandra

vsethia-adi, 2021-12-02:
Hi Alexandra,
Attached code is working for 0.4.0 but not for 0.5.0.

First spi_write_and_read function executes without any issues. But the next one does not. It hangs indefinitely.

Regards
Vikash

[External]

AlexandraTrifan, 2021-12-02:
I'm sorry, but I can't see any attachment to your reply.

vsethia-adi, 2021-12-02:
Attaching it as a text file.

[External]

# This is a sample Python script.

# Press Shift+F10 to execute it or replace it with your code.
# Press Double Shift to search everywhere for classes, files, tool windows, actions, and settings.

from sys import exit
import libm2k
import time
# import matplotlib.pyplot as plt
import bokeh
from bokeh.plotting import figure, output_file, show
from bokeh.models import LinearAxis, Range1d
from bokeh.models import HoverTool, BoxSelectTool #For enabling tools
import pandas as pd
import numpy as np
import struct
NB_OUT_SAMPLES = 1024

DIG_CH_SCLK = libm2k.DIO_CHANNEL_0
DIG_CH_MISO = libm2k.DIO_CHANNEL_1
DIG_CH_MOSI = libm2k.DIO_CHANNEL_2
DIG_CH_CSB = libm2k.DIO_CHANNEL_3
DIG_CH_BUSY = libm2k.DIO_CHANNEL_4
DIG_CH_TGP = libm2k.DIO_CHANNEL_5
DIG_CH_CONVST = libm2k.DIO_CHANNEL_6
DIG_CH_LDACB = libm2k.DIO_CHANNEL_7
DIG_CH_CMP = libm2k.DIO_CHANNEL_8
DIG_CH_RESETB = libm2k.DIO_CHANNEL_10
DIG_CH_ALERT = libm2k.DIO_CHANNEL_11
DIG_CH_TESTIN = libm2k.DIO_CHANNEL_12

ANALOG_IN_CH_0 = libm2k.ANALOG_IN_CHANNEL_1
ANALOG_IN_CH_1 = libm2k.ANALOG_IN_CHANNEL_2

POWER_PORT_0 = 0
POWER_PORT_1 = 1

BUFFER_SIZE = 10000
SAMPLES_PER_PERIOD = 16
OFFSET = 5

SAMPLING_FREQUENCY_IN = 1000000
SAMPLING_FREQUENCY_OUT = 10000
OVERSAMPLING = 1

def main():
    context_vec = libm2k.getAllContexts()
    if (len(context_vec) <= 0) :
        print('Connection Error: No ADALM2000 device available/connected to your PC.')
        exit(1)
        
    context = libm2k.m2kOpen(context_vec[0])
    # context = libm2k.m2kOpen('ip:192.168.2.1')
    if context is None:
        # context = libm2k.m2kOpen('usb:3.7.5')
        # context = libm2k.m2kOpen('usb:1.6.5')

        # if context is None:
        print('Connection Error: No ADALM2000 device available/connected to your PC.')
        exit(1)
        # else:
        #     print('ADALM2000 connected :' + context.getUri())
    else:
        print('ADALM2000 connected :' + context.getUri())

    # check if mixed signal is available on your firmware version
[... 139 lines cut ...]
        print('SPI Error: Could not transmit the data')
        exit(1)

    

    # DAC Data registers of E4.MIR.VD[1:0], E5.SPH.VD, E12.WHTR.VD, E9.RFPD.VD
    data = bytearray([0x80, 0x3A, 0x00, 0x00, 0x80, 0x3C, 0x00, 0x00, 0x80, 0x4E, 0x00, 0x00, 0x80, 0x60, 0x00, 0x00, 0x80, 0x64, 0x00, 0x00 ])
    status = libm2k.spi_write_and_read(spi_desc, data)

    #buffer = libm2k.spi_create_buffer(spi_desc, data)
    #status = libm2k.spi_write_and_read_samples(spi_desc, buffer, data)
    if (len(data) == 0) or (status != 0):
        print('SPI Error: Could not transmit the data')
        exit(1)
    print(data) 
    
    
    libm2k.spi_remove(spi_desc)
    libm2k.contextClose(context, True)

# Press the green button in the gutter to run the script.
if __name__ == '__main__':
    main()

# See PyCharm help at https://www.jetbrains.com/help/pycharm/

AlexandraTrifan, 2021-12-02:
Thank you!
I'll run some tests and get back to you!

-Alexandra

AlexandraTrifan, 2021-12-02:
You are using the following setup: `m2k_spi_init.cs_polarity = libm2k.LOW_LEVEL_DIGITAL`
Please initialize the cs_polarity attribute with one of the following values:` cs_polarity - {ACTIVE_LOW | ACTIVE_HIGH}` and let us know if this fixes the issue for you.

-Alexandra

vsethia-adi, 2021-12-08:
Hi Alexandra,
Yes, this fixes the issue for me.

AlexandraTrifan, 2021-12-08:
Happy to hear this!
Closing this issue, please let us know if you encounter any other issues with libm2k!

-Alexandra

closed 2021-12-08 by AlexandraTrifan