analogdevicesinc/libm2k #295: libm2k Runtime Error in Raspbberry Pi

codenio, 2021-12-02:
I am working with two ADALM2000 units using `concurrent.futures.ProcessPoolExecutor`in python. 
It works as expected in Ubuntu-18.04 without runtime errors 

When used in Raspberry Pi, ADALM gets ejected and libm2k fails with the following errors

[image]

[image]

context :
- I had installed libiio using master_latest_libiio-raspbian-8-armhf.deb package
- I had built python binding for libm2k from the source
- It works as expected for a single ADALM device

would like to understand this error, could someone help.?
thanks in advance

rgetz, 2021-12-02:
Do you have any other usb devices plugged in at the same time? I think there is a limit on the different Pi’s on the number of end points - Pi 4 is 32 endpoints. Each M2k is ~5.

codenio, 2021-12-03:
thanks for the prompt response.
I am using Pi 4 Model B and It is connected to two units of ADALM connected via USB.

codenio, 2021-12-11:
can someone help me by providing pointers for debugging the issue..?

codenio, 2021-12-11:
while using two m2k devices concurrently for 4 channel data acquisition with 2 channel signal generation (one DAC in each m2k)

This error occurs when **ctx.calibrateADC()** is called the second time for the 2nd device while keeping the 1st device open

[image]

4 channel data acquisition fails when used with signal generation and works concurrently as expected (Without signal generation)
 
[image]

is this a bug...? can someone help me by providing pointers to debug and fix this issue..?
thanks in advance
codenio

AlexandraTrifan, 2021-12-13:
Hi,

You can save the state of the board before the first call and before the 2nd call using iio_info, so we can check for differences between the 2 log files. Use` iio_info -u "usb or ip uri here" > before.txt` and attach the text files here please. Using them we can establish if there is some attributes with weird values. 

Are you using oversampling_ratio or any triggers in the ADC? You could try to add a "reset" call before running the ADC and DAC calibration in you scripts - this will prevent any bad initial configuration in the ADC or DAC. (example: https://github.com/analogdevicesinc/libm2k/blob/master/bindings/python/examples/analog.py#L46-L47 ).

Could you send us an example script? It would be much easier to reproduce the issue and check out what's wrong.

Thank you!
-Alexandra

codenio, 2021-12-14:
Thanks @AlexandraTrifan for the prompt response, 

Attempted exporting params before the second call, but it fails with the following info
```shell
$ iio_info -u 'usb:1.5.5' > before_2.txt
ERROR: Unable to claim interface 1:5:5: Device or resource busy (-16)
Unable to create IIO context usb:1.5.5: Device or resource busy
```

PFA the sample code to replicate this issue

```python
import libm2k
import time

# populate the devices list with in the order that needs to be called.
devices = ["****4846", "****b809"]

def mapDeviceURIs(devices):
    lst = libm2k.getContextsInfo()
    devicesMap = {}
    for i in range(len(lst)):
        if lst[i].serial in devices:
            devicesMap[lst[i].serial] = lst[i].uri
    return devicesMap

device_map = mapDeviceURIs(devices)

print(device_map)

process_params = []
for serial in devices:
    # Generate Process Params Dynamically
    process_params.append({"device": serial,
                        "uri": device_map[serial]})
print(process_params)

for params in process_params:
    
    print(f"Starting: Attempting to open {params['device'][-4:]} device...")
    ctx=libm2k.m2kOpen(params['uri'])
    print(f"Connecting M2k @ {params['uri']}")

    if ctx is None:
        print("Connecting Error: Unable to Open M2k")
    else:
        # get analog input object
        ain = ctx.getAnalogIn()
        # create signal generator
        sgen = ctx.getAnalogOut()
       
        # Prevent bad initial config for ADC and DAC
        ain.reset()
        # Prevent bad initial config for ADC and DAC
        sgen.reset()
        time.sleep(2)
        
        print("Calibrating DAC")
        # calibrate DAC
        ctx.calibrateDAC()
        time.sleep(2)
        
        print("Calibrating ADC")
        ctx.calibrateADC()
        time.sleep(2)
        params['ctx'] = ctx
        cmd = input("Enter c to close m2k : ")
        
        if cmd == "c":
            libm2k.contextClose(ctx)
            print(f"Closed M2k @ {params['uri']}")
        else:
            print(f"Keeping M2k @ {params['uri']} open")
        
print(params)
```

AlexandraTrifan, 2021-12-15:
Hi,

I used the command **dmesg** to check what's happening to the USB devices and why they get ejected once the calibration starts for the second M2K, while the first one is still connected. In the output log I found the following:

> [ 9436.191527] usb-storage 1-1.3:1.2: USB Mass Storage device detected
[ 9436.191772] scsi host1: usb-storage 1-1.3:1.2
[ 9436.193043] cdc_acm 1-1.3:1.3: ttyACM1: USB ACM device
[ 9437.207576] scsi 1:0:0:0: Direct-Access     Linux    File-Stor Gadget 0504 PQ: 0 ANSI: 2
[ 9437.207908] sd 1:0:0:0: Attached scsi generic sg1 type 0
[ 9437.210515] sd 1:0:0:0: [sdb] 61441 512-byte logical blocks: (31.5 MB/30.0 MiB)
[ 9437.210812] sd 1:0:0:0: [sdb] Write Protect is off
[ 9437.210821] sd 1:0:0:0: [sdb] Mode Sense: 0f 00 00 00
[ 9437.213435] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[ 9437.249931]  sdb: sdb1
[ 9437.253273] sd 1:0:0:0: [sdb] Attached SCSI removable disk
[ 9462.091240] usb usb2-port1: over-current change 
[ 9462.327339] usb usb2-port1: over-current condition
[ 9462.327454] usb usb2-port2: over-current change 
[ 9462.328712] cdc_ncm 1-1.4:1.0 usb0: unregister 'cdc_ncm' usb-0000:01:00.0-1.4, CDC NCM
[ 9462.567462] usb usb2-port3: over-current change 
[ 9462.807585] usb usb2-port4: over-current change 
[ 9468.155020] rndis_host 1-1.3:1.0 eth1: unregister 'rndis_host' usb-0000:01:00.0-1.3, RNDIS device
[ 9468.264488] usb 1-1-port1: over-current change 
[ 9468.368514] usb usb2-port3: over-current change 
[ 9468.497937] usb 1-1-port2: over-current change 
[ 9469.217958] usb 1-1.3: USB disconnect, device number 30
[ 9469.227844] scsi 1:0:0:0: rejecting I/O to dead device
[ 9469.351604] usb 1-1-port4: over-current change 
[ 9469.517848] usb 1-1.1: reset full-speed USB device number 29 using xhci_hcd
[ 9473.937493] usb 1-1.4: new high-speed USB device number 31 using xhci_hcd
[ 9474.079347] usb 1-1.4: New USB device found, idVendor=0456, idProduct=b672

Can you check whether you have the "over-current condition/change" errors appearing on your RPi when the ADALM2000 gets disconnected?

This happens because the RPi cannot support both ADALM2000s and in order to fix this you need to to connect at least one ADALM2000 using an external power supply or connect both ADALM2000s into an externally powered USB hub.

-Alexandra

codenio, 2021-12-15:
@AlexandraTrifan thanks for the support, 

> This happens because the RPi cannot support both ADALM2000s and in order to fix this you need to connect at least one ADALM2000 using an external power supply or connect both ADALM2000s into an externally powered USB hub.

Initially, I had this same doubt (though I did not check the log) and expected a low current warning to pop on UI, but to my surprise, I did not receive one.

I had tried powering both m2k via separate cables from raspberry pi USB ports themselves and this issue persisted.
let me check the log and get back. 
Will try powering up m2k via an external power supply instead of tapping again from raspberry pi.

codenio, 2021-12-16:
yes, it is due to over-current change, powered both m2k via the external power supply, It works as expected.
@AlexandraTrifan  Thanks for the support. :handshake:

closed 2021-12-16 by codenio