analogdevicesinc/libm2k #297: Modifying the status of unused DIO pins after configuring SPI or I2C interfaces in the python bindings.

vsethia-adi, 2021-12-17:
Can you please add the following features where I can concurrently use the SPI / I2C interface functions and also modify the other DIO pins which can change states during a SPI or I2C functions..

vsethia-adi, 2021-12-27:
Also, if I do not wish to use the SPI/I2c libm2k functions, but a generic stream of binary data, I wish to know how to capture the incoming data whether serial or parallel.

teoperisanu, 2022-01-03:
We have been working on adding this functionality. After several tests it seems that it works as expected. Can you install this version: 
- https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/42065600/artifacts
and confirm that it also works for you. Please make sure to install only one Python binding (the one you are using) and also install libm2k tools:
[image]

An example of how to use the new API can be found here: https://github.com/analogdevicesinc/libm2k/blob/expose_spi_create_buffer/bindings/python/examples/expose_spi.py

The current modifications are applied only for SPI, but after making sure that this is what you want, we will add it also to the other protocols.
If you run into any troubles please let us know.

- Teo

vsethia-adi, 2022-01-12:
This method is working for me.

teoperisanu, 2022-02-02:
Both the fix for SPI getting stuck and the new API (for creating an SPI buffer) are on master.
The artifacts will be soon available here: https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/42427793/artifacts

closed 2022-02-02 by teoperisanu

vsethia-adi, 2022-02-03:
Hi Teo,
Is the SPI buffer getting modified after read?
I wish to access the SPI buffer during both write and read and verify all the I/Os. Let me know if that is feasible.

vsethia-adi, 2022-02-10:
If I have to use an external m2k, what is the set of IP addresses I can use apart from 192.168.2.1.?

Can you provide me access to the method that converts the acquired buffer memory to data variable for SDO. If I use the digital.push() method to transfer the buffer, I should be able to use the startAcquisition() and getSamples() to get the acquired buffer memory.  I believe that you might be using the same or similar function to acquire and extract data for SDO. Hence the request,

teoperisanu, 2022-02-10:
Instruction for using multiple devices: https://wiki.analog.com/university/tools/pluto/users/customizing#multiple_devices
Basically you have to change only the ipaddr from config.txt (let's say from 192.168.2.1 to 192.168.2.3) and then eject the board. If the ip doesn't change you need the unplug and replug the board.
I have exposed the processSamples method. Parameters:
- desc: SPI descriptor
- data: an array of fixed size where the human readable data will be stored after processing
- bytesNumber: the size of data
- samples: the digital buffer containing the samples to be process

The artifacts will be soon available here: https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/42520340/artifacts
Can you try this version and let us know if it works for you.

Thank you,
Teo