analogdevicesinc/libm2k #298: libm2k spi_write_and_read() function getting stuck when called multiple times

vsethia-adi, 2022-02-01:
I am trying to call the spi_write_and_read() function multiple times to evaluate the functionality of the device. and I have observed that the call to the spi_write_and_read function is getting stuck at random times depending on the clock speed. 
Is there any timeout function to abort the spi_write_and_read function?
Is there a way to know whether the write has happened and it is the read that is pending? as I do not see any issue with the spi_write_only function.

Can somebody have a look at it and let me know of any workaround?
I have shared the code in email.

teoperisanu, 2022-02-01:
Hi,

can you try this version: https://ci.appveyor.com/project/analogdevicesinc/libm2k/builds/42411163/artifacts.
In this libm2k version the reading and writing threads are a bit more synchronized.

Thank you,
Teo

vsethia-adi, 2022-02-01:
Hi Teo,
Thanks. This version fixes the issue.
Shall I assume that this version has the spi_write_and_read_samples with buffer option to modify the non-SPI pins?

vsethia-adi, 2022-02-02:
Is there any control over the data read position on MISO. ?
The device is capable of running at 40MHz sclk with clk to miso delay of 10ns max. I noticed that with 10MHz clock, there is no issue regarding the data capture on miso. whereas at 20MHz, the miso data is shifted by one bit. Hence the query..

teoperisanu, 2022-02-02:
You cannot shift the MISO data. Even though the device supports 40MHz, I would recommend you not to exceed 25 MHz. 
The problem you encountered is most likely to be due to a miscalculation - not sure if it's a bug or not, but I will investigate.
Can you try with 25MHz and tell us if you are still experiencing the same issue?

adisuciu, 2022-02-02:
The sample rate of the device is 100MSPS. With our implementation you can only get clock speeds which are integer dividers of this 100MSPS. 40MHz is not an integer divider therefore if you generate clock at that frequency, the clock will not have an equal number of samples on HIGH/LOW. The algorithm kinda relies on this fact. You can try providing external clock if you need 40MHz and generate SPI communication from that.

vsethia-adi, 2022-02-02:
Hi Adrian,
I am aware of the master clock of 100MHz. 
That is why, I am trying at 20MHz, which is a integer divider of the master clock.  

Hi Teo,
It works at 25MHz, and I do not see the same issue. I tried again at 20MHz, and it does not work. There is a bit-shift.

teoperisanu, 2022-02-02:
The result of division between the master clock and the clock in use should be an even number, because each clock cycle is composed of an even number of samples.

vsethia-adi, 2022-02-02:
Seems a valid explanation. Thanks.

closed 2022-02-02 by vsethia-adi