analogdevicesinc/libm2k #342: How to set sampling frequencies <100k?

Modiug, 2023-09-12:
I entered the same issue in [gr-m2k](https://github.com/analogdevicesinc/gr-m2k/issues/28). I guess this applies to libm2k as well?

When I set a "Sampling frequency" of 10k, I get this:
"RuntimeError: ERR: Runtime - Device: Cannot set the number of kernel buffers"

I tried different "Buffer sizes" and "Kernel buffers", but to no avail.
I was able to set "Sampling freqency" to 100k and "Oversampling ratio" to 8, which gives an effective sample rate of 12.5k. But using an "Oversampling ratio" of 10 again leads to the above error message.

Could somebody point out the limits and relationsships between "Sampling frequency", "Buffer size" and "Kernel buffers"?

NanoExplorer, 2024-01-26:
I think that this is more of a gnuradio issue than a libm2k issue, since I can set these values just fine with the python bindings of libm2k.
```python
m2k = libm2k.m2kOpen()
ain = m2k.getAnalogIn()
ain.setKernelBuffersCount(10)
ain.enableChannel(0,True)
ain.enableChannel(1,True)
ain.setSampleRate(100000)
ain.setOversamplingRatio(10)
```
All of these run just fine for me, again on Python. It would help the folks over at gr-m2k if you shared your code (or a screenshot of the settings dialog in gnuradio) so that they might be able to reproduce the issue...

I take it you've already tried the settings listed here: https://wiki.analog.com/university/tools/m2k/libm2k/gr-m2k and only changing the sampling frequency?

Modiug, 2024-01-26:
I can confirm that your code snippet works for me.
However GNU Radio Companion from radioconda-2023.11.06-Windows-x86_64 has still the same issue.
I'll leave a comment with the gr-m2k team.

adisuciu, 2024-01-26:
This will be fixed in the latest version of libm2k. We're preparing a release asap.

Modiug, 2024-01-26:
Oh, ok. I understood from @NanoExplorer it could not be reproduced in libm2k directly.
Anyway, thanks a lot!

AlexandraTrifan, 2024-02-01:
The latest libm2k release v0.8.0 was published.  https://github.com/analogdevicesinc/libm2k/releases/tag/v0.8.0
This release fixes the above mentioned issue.

closed 2024-02-14 by adisuciu