analogdevicesinc/libm2k #352: M1 Mac platform not supported

Jnesbeth, 2023-12-04:
```
┌─────────────────────────────────────────────────────────────────────────────────────────┐
│  SYSTEMINFORMATION                                                     Version: 5.21.20 │
└─────────────────────────────────────────────────────────────────────────────────────────┘

Operating System:
──────────────────────────────────────────────────────────────────────────────────────────
Platform         : darwin
Distro           : macOS
Release          : 14.0
Codename         : macOS Sonoma
Kernel           : 23.0.0
Arch             : arm64
Build            : 23A344

System:
──────────────────────────────────────────────────────────────────────────────────────────
Manufacturer     : Apple Inc.
Model            : MacBookPro18,1
Version          : 
Virtual          : 

CPU:
──────────────────────────────────────────────────────────────────────────────────────────
Manufacturer     : Apple
Brand            : M1 Pro
Family           : 458787763
Model            : 
Stepping         : 4
Speed            : 2.4
Cores            : 10
PhysicalCores    : 10
PerformanceCores : 8
EfficiencyCores  : 2
Processors       : 1
Socket           : SOC
```

Python Version: Python 3.10.11

Hello, 

I was trying to install the libm2k library on my macbook for a project I was working on. 

Installation logs using pip:
```
Jason.Nesbeth@JNESBETH-M01 ~ % python3.10 -m pip install -v -i https://test.pypi.org/simple/ libm2k 
Using pip 23.3.1 from /Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pip (python 3.10)
Looking in indexes: https://test.pypi.org/simple/
ERROR: Could not find a version that satisfies the requirement libm2k (from versions: none)
ERROR: No matching distribution found for libm2k
```

From what I can tell there is no platform support for my flavor of Mac at the moment with arm64 architecture. I was wondering if there was going to be at some point?

Thanks!

AlexandraTrifan, 2023-12-07:
Hi,

Right now we don't have Continuous Integration images that run Mac arm64, so we don't have continuous builds for libm2k artifacts for arm64. 
We plan on adding these images and once we do, we will also build pip packages for arm64. I could not provide an exact timeline for this, it will probably be in the first months of next year. 
Until artifacts/installers are provided, you could build the library and python bindings from source, if that is an option for you.

Thank you!
-Alexandra

NanoExplorer, 2024-01-26:
Also the Anaconda installer works just fine for me on arm64 mac.

AlexandraTrifan, 2024-02-01:
> Except for the fact that closing the device generates an error. When I call `libm2k.contextClose(ctx)`, it gives the error message: `m2k-fabric has no user_supply_powerdown attribute for the selected channel`. It seems like it does close the device correctly however, because I can immediately reopen it and everything still works.

I managed to reproduce this error message with the following scenario:
-ADALM2000 firmware version v0.32
-libm2k v0.7.0
-libiio v0.24

There is a compatibility issue between firmware v0.32 and libiio v0.24.
I would recommend updating libiio to v0.25 and libm2k to v0.8.0 if possible. If you are using Conda packages, the libm2k Conda package for v0.8.0 should be available in the next days (an update gets automatically generated when we publish the release).

If updating is not a possibility, then you could use firmware v0.31 for the moment (but keeping in mind that v0.32 has some useful bug fixes for the M2K).

-Alexandra

closed 2024-02-14 by adisuciu